cleanflight/cleanflight-configurator #34: Latest CF GUI not saving Modes to Naze32

Hands0n, 2015-01-02:
From one of our forum members:

"cleanflight cant save modes to my full naze32. Save to the eprom in cleanflight then when I disconnect and reconect one of the flight modes are saved."

I've seen similar reported elsewhere also, but not on here. So here's a potential bug report.

Has something changed or is this a bug?

hydra, 2015-01-03:
please provide steps to repeat.  Not have any issues here.  Make sure you're running latest stable chrome, firmware, and configurator.

hydra, 2015-01-07:
Without any further information I can't progress this issue.

Closing the issue.

If you can provide the additional requested information it can be re-opened and investigated.

closed 2015-01-07 by hydra

Hands0n, 2015-01-07:
Hi Dominic

Will run my own tests and see if I can replicate the symptoms. If i can
then will post details and video for you to review.

Regards
Danny