cleanflight/cleanflight-configurator #35: WS2812B with CleanFligh

maverikt, 2015-01-03:
Hi,
I am using WS2812B which were purchased here:
http://www.ebay.co.uk/itm/1m-WS2812B-RGB-LED-Pixel-Strip-60-Individually-Addressable-Pixels-60-m-/151478784787?

I have made a test strip and connected as specified in the Wiki.  However the LED are behaving randomly, incorrect colours and incorrect LED lit.  You mentioned recompiling the source with a different frequency but ive not managed to successfully re compile the source, would it be possible to add the frequency to the CLI tools please.

Has any one successfully got the WS2812B's to work. 
Also can anyone explain how the ID of each led is set, I know its different but before ive used an I2C bus but each device had to have its ID set.  How can i know that my first LED in the strip is the first ID???

Many Thanks,
Tony Brock

kylemanna, 2015-01-03:
Documentation @ https://github.com/cleanflight/cleanflight/blob/master/docs/LedStrip.md

Existing issue @ https://github.com/cleanflight/cleanflight/issues/196

Do you have a 3.3V -> 5V level converter?  See issue cleanflight #196 for details.

maverikt, 2015-01-03:
Hi,
Unfortunately I dont have a 3.3v -> 5V level converter.  The datasheet for my LED's is here:
http://www.lightgear.co.uk/WS2812B.pdf
Tony

kylemanna, 2015-01-03:
>Unfortunately I dont have a 3.3v -> 5V level converter. The datasheet for my
>LED's is here:
>http://www.lightgear.co.uk/WS2812B.pdf

Datasheet is the same as the datasheet referenced in the other thread.
The specs require a level converter.  See the other issue for more
discussion.

maverikt, 2015-01-03:
Hi,
I have found i have one of these, sparkfun level converter documented here, can you advise how i should wire it if i can use it please.
https://www.sparkfun.com/products/12009
Many Thanks for any time you can give to help me,
Tony

hydra, 2015-01-07:
There are two problems here:
1) This issue has been raised in the wrong issue tracker, it's firmware related not configurator related.
2) The issue tracker is not for support or help, it is for reporting bugs and feature requests.

Please ask on IRC or the forums for assistance.

Closing this issue.

closed 2015-01-07 by hydra

maverikt, 2015-01-07:
Hi,
Just to let you know i upgraded to the latest firmware and the 2812B's works just fine with no level converter.

Tony