cleanflight/cleanflight-configurator #36: Automatically pause between lines in the CLI to prevent dataloss

thenickdude, 2015-01-03:
When pasting a complete dump into the CLI, data is lost because the Naze can't keep up with the writes (I tend to lose the AUX configuration from my dumps). Adding a delay between each line sent on the serial bus should cure this and make restoring dumps easier for everybody.

Exidouce, 2015-01-10:
You could also lower the baud rate. As a temporary alternative. That should help with the data processing.

andymac4182, 2015-02-09:
I believe the timeout just needs to be set here https://github.com/cleanflight/cleanflight-configurator/blob/master/tabs/cli.js#L40 It appears to support sending slowly now but is set to send with no delay.

michelp29, 2015-04-04:
I'm not sure this is neither a timeout issue, nore a large-number-of lines-issue, because I can reproduce the specific "aux" problem with just a few "aux" lines in the CLI : have a look at issue #154
In my mind it seems to be a specific "aux" command issue.

thenickdude, 2015-04-05:
Michelp29, as far as I know the problem is actually the "profile 0" command, which takes so long to process that the "aux" commands which follow it are lost.

hydra, 2015-05-17:
It's not specifically the aux command, i get issues when pasting in a large number of 'set' commands on some boards too.

hydra, 2015-10-14:
I think we should not do this.  Development effort is better spent elsewhere, the workaround is known and as I have mentioned tools should not attempt to use the CLI.  It's meant for humans.

Yes I acknowledge it's a bit of a pain.

Related issues

https://github.com/cleanflight/cleanflight/issues/325
https://github.com/cleanflight/cleanflight/issues/1332

closed 2015-10-14 by hydra