cleanflight/cleanflight-configurator #37: 0.58.0 just 'waiting for data'

youvebeenwarned, 2015-01-04:
Hello,

It seems cleanflight auto updated to ver 0.58.0 and now I cannot connect to my naze32 FC for further tuning. It just sits on the 'waiting for data' animation and doesn't go any further.

Any clues as to whats up? I'm running on OS X 10.8.5, and all was working fine until this update from the 3rd Jan.

Thanks, I hope you can help.

Roeland54, 2015-01-04:
You just need to update the firmware of the naze to the latest version.

youvebeenwarned, 2015-01-04:
LOL, I just saw that I wasn't actually running .58 ver, but 57, it seems it didn't update. Anyway I've just uninstalled and installed again and its now the latest version.

I do now have to configure the board again to all my previous settings right? I've saved a config file previously but I still have to configure switch etc. Is it not possible to make it so that we don't have to do so much reconfiguring every time it updates? Seems a bit crazy and counter productive.

Thanks

Roeland54, 2015-01-04:
This issue https://github.com/cleanflight/cleanflight-configurator/issues/23 already recommends a solution for this problem.

youvebeenwarned, 2015-01-04:
Ok, so wait, now I want to reconfigure using my previously saved config file (that was becoming a real nice set-up for me) but can't because get the message " Backup file provided was generated for older version of configurator and is incompatible with this version of configurator. Sorry".

So now I've got no access to my previous settings, the settings that where very close to being perfect for me. How is this progress?

youvebeenwarned, 2015-01-04:
Hi Roeland54, so your saying I can 'dump' my now incompatible config file from a few days ago (pre latest update) onto the board useing CLI? How do I do that?

Thanks all.

youvebeenwarned, 2015-01-04:
And another thing, where's the option to turn off auto update? I was getting along just fine and making some real progress and now I've got to re-config all over again. If update had asked me 'hey there is an update, should I go ahead and update for you', I would have said 'No thank you, every thing is just fine for me as it is'

Sorry, but this stuff is just really frustrating.

hydra, 2015-01-04:
I feel your pain. It would be nice to devote more time to making the configurator more backwards compatible with older versions.

The installation.MD in the cleanflight docs details CLI backup/restore process.

There is a feature request to allow CLI access when incompatible api changes prevent gui usage/restoration.

Old firmware and old configurator versions are archived on the releases page of each project.  You can choose to use which ever one you like to allow you to make CLI backups.

See chrome documentation for auto update.

closed 2015-01-04 by hydra

youvebeenwarned, 2015-01-04:
Thanks Hydra, 

Can I not just simply load the previous version from the chrome store. Or download an older version elsewhere. If so how do I install on OS X?

youvebeenwarned, 2015-01-04:
Also do you have a link to the 'releases page' you mention? I'll admit I'm having a hard time navigating this site.

youvebeenwarned, 2015-01-04:
looking at it some more, it seems only the 'current' version is every available via the chrome store. 

So my question really is how do I physically roll back to a previous version of cleanflight on my Mac. I am in no way a programmer.

KyotoFox, 2015-01-04:
@youvebeenwarned Download v0.57.2 from here and unzip it: https://github.com/cleanflight/cleanflight-configurator/releases
Follow this guide from step 5 to start it: https://developer.chrome.com/apps/first_app

youvebeenwarned, 2015-01-04:
Is there a version previous to this? I think its this version that started it all. I'll give it a try now though anyway. Thanks

KyotoFox, 2015-01-04:
You can select an older commit and download the code at that point from here: https://github.com/cleanflight/cleanflight-configurator/commits/master
Click "<>" on a commit, then you can download a zip-file on the right.

youvebeenwarned, 2015-01-05:
Ok, I've downloaded the 'Zip', now what do I do with it. The 'read_me' file doesn't open on my Mac.

Thanks

hydra, 2015-01-05:
You can view the readme file with any text editor. Please seek further assistance on the forums or irc.  

The issue tracker is for reporting bugs and suggestions and not for user support.