cleanflight/cleanflight-configurator #40: Consider backwards compatibility when creating new versions

nebbian, 2015-01-06:
An issue that is quite frustrating to me (and others) is the way that occasionally, when a new version of the configurator is released, it breaks all communication with boards that are running the previous version of Cleanflight.

This has happened a number of times in the past, and most recently with the version that was released around the 5th January 2015.

The effect this has is that the configurator will silently update itself, so when the user goes to adjust some PID values, or change the way that aux channels are configured, the configurator can't connect to the board.  This happens without the users knowledge in most cases, so it's a complete surprise when the user is locked out of being able to fly their craft.

What would be great is if some consideration is made towards backwards compatibility when changes are made.  This would mean that users would be given some warning before being locked out of being able to use this software.

michelp29, 2015-01-06:
This would be great, and should avoid lots of user frustration using Cleanflight.

It seems you guys are making great job :) and I thank you for this;
frequent update is good news, but should be done with a minimum backward compatibility.

kylemanna, 2015-01-07:
I've been burned by this too

Exidouce, 2015-01-07:
While I have not been burned by the forced update (yet) I agree with this 100%

strips, 2015-01-08:
Personally I would not like the project to have to consider backwards compatability for every change. This will slow down new development considerable! 

But maybe doing things a bit different might be an acceptable solution. 

Suggestions :
- Having auto-backup with a new restore functionality that highlights unknown or out of bound values. 
- Be able to access CLI or an backup function (dump) without a full connect. 

Regards 
Stian

michelp29, 2015-01-08:
Yes, I agree that a full backward compatibility is time-consuming and not really necessary.
But at least having a degraded-mode on the GUI, allowing to just connect and let the user doing a last backup and CLI on the FC, before the user upgrades the firmware.
For the last one (letting CLI on previous fw version) I've seen this in the todos :)

nebbian, 2015-01-08:
https://github.com/cleanflight/cleanflight/blob/master/src/main/io/serial_msp.c#L98-L99

hydra, 2015-01-08:
Latest configurator has config migration and CLI access as a backup when api versions are incompatible.

Closing this as fixed :smile:

closed 2015-01-08 by hydra

michelp29, 2015-01-08:
Thanks, I've tested and it works as expected :-)

dogz85, 2015-01-20:
not fixed, this is not compatibility, just a way to migrate settings.

at the least the user should have access to PID's

Exidouce, 2015-01-20:
You can use an old configurator. Part of the issue is google does not allow you to stop auto update. There are a couple places that tell you how to run an old version.

I think being able to migrate settings is great. That did not used to exist. You have to balance progress with backwards compatibility. With changes that are being made(big ones) you cannot just assume backwards compatibility.

dogz85, 2015-01-20:
I dont think that progress at the loss of usability is a great goal. this forced updating is affecting the whole userbase. what do you think the average low tech user is going to do? they log into the computer connect up the copter and wonder why its broken.  then they blame cleanflight and go back to what they know works. 

even stacking the apps and having an option (dropdown within the app)  to select the matching the GUI to the firmware. sure it will bloat the app but it would work. well to a point. atm cleanflight is moving so fast with so many changes their would be 10 or 15 versions. 

or build the app using something other than chrome.