cleanflight/cleanflight-configurator #43: documentation: correct COM port must be selected for flashing

Philippe-France, 2015-01-06:
It's not mandatory to be connected to COM port to flash via USB, but correct COM port must be selected in top left field otherwise flash is not starting (sorry I don't note the error message).
Release 0.59.0
/Phil

hydra, 2015-01-07:
Which documentation were you referring to?

Philippe-France, 2015-01-07:
I do not refer to any doc as I don't find it, but just creating a ticket as reminder for doc writer about the behavior I observed.

hydra, 2015-01-07:
Ahh OK. I'm not sure what to do with this ticket since there is not current plan to document the configurator.  The idea is that it should be simple enough Si that it dies not require documentation.

There was an idea yesterday about having modes for the configurator: flashing, configuration.  The ui would be updated to make it more obvious.

I will close this ticket since there is nothing to do.

closed 2015-01-12 by hydra