cleanflight/cleanflight-configurator #46: CleanFlight logo

Dvogonen, 2015-01-07:
I noticed that the CleanFlight configurator is using the Baseflight icon, although with another color. I also noticed that there seems to exist no common logo or graphical profile for CleanFlight neither on the home page, on github or in the configurator itself. Logos and graphical profiles might seem like trivial details, but are important for the recognition and perseption of a open source "brand". High quality software is always branded by some sort of visual implements.

I played with the letters C and F in CleanFlight and made a few scetches of what could be used as both logos and icons. I am not a graphical artist and know that others are far better at this than I am My attempts may perhaps serve as an inspiration for someone else.

[image]
[image]
[image]

hydra, 2015-01-07:
Thanks for these.

Not sure if you're aware but there is a competition going on right now to design a new logo, the details are here:

https://www.facebook.com/permalink.php?story_fbid=623759171067204&id=582474325195689

We definitely want to stick with Green and need them in SVG format.

Any chance you could update them as per the competition details?

Dvogonen, 2015-01-07:
Had no idea there was an ongoing competition.
I will paint the logos green and enter them.

closed 2015-01-07 by Dvogonen