cleanflight/cleanflight-configurator #55: prop balance settings

handsomejackuk, 2015-01-14:
This is something i seem to use quite a bit.. lately.. the motors config Tab its great... but would be even better, super, fab, fantastic better.... :-) if there was an option maybe to set some. fixed motor speeds for balancing... 

so it would work like this.... set a speed 1300 maybe. then have the ability to quickly switch from motor 1 to 4 with the same speed setting....the ability to log the acc data too while doing these quick tests would show which motor or prop needs the most balancing..

I just find it a very manual way to do it at the moment... obviously it would need some very strict warnings when props were attached...!!!!!!

OOOOOOooooohhhhh OOOOOoooouch....

or some other automated method.of getting a consistent result...

I realise that the method of balancing the props is not going to change but at least we can all see the results far easier..

Thanks Hydra and all the rest of the team for all the hard work you putting in.. I love Cleanflight and still continue to spread the word... great work fellas...

vadav, 2015-01-18:
Could be done when using the master, separate motors cant be adjusted higher than what the master is set to.

fiveangle, 2015-02-06:
Having a binary ON/OF toggle check box widget below each motor level graph that acts if it is in "series" with the current master motor control would resolve (with an "all" override which grays the individual switches out when engaged if you wanna be all fancy :)
\
-=dave

andymac4182, 2015-02-09:
Is this for a particular type of prop? I was under the impression a prop balancer was fairly easy way to balance props?

fiveangle, 2015-02-09:
Statically balancing props in the 6" and less is not useful. Dynamic laser balancing is required if you want to squeeze the maximum performance out.

Here's a rudimentary example of how it works: http://youtu.be/PqtUSSdf8b4

-=dave

Ozzy33, 2015-03-19:
would be sweet to record the max (rms) vibration values (one 3 axis) to compare if your getting better or worse (like a data log)

cmavr8, 2015-04-17:
I also think that:
a. dynamic balancing is very important (fighting jello)
b. doing it using the configurator is the most efficient way, and currently one of Naze's best features IMO. On my APM tricopter I have to resort to cumbersome methods and I hate it.
c. having and easy way to test the motor/prop at constant, predefined RPM is crucial. I balance mine at hover-throttle
d. having an easy way to measure vibrations overall and compare to previous settings (without having to click on reset with the mouse) would be very nice to have

I suggest that we formalize the above, and put them in two new feature requests, then close this. 
Does anyone agree? Should I start? :)

Ozzy33, 2015-04-18:
I do, YES.

cmavr8, 2015-06-26:
Damn, it took me 2+ months to get to it. Anyway here it is:

Usecase:
Bob wants to dynamically balance his multicopter. He ties it down, removes the props, (starts his Tx,) connects LiPo power and finally plugs the USB cable to his PC. He starts Cleanflight-configurator, connects it (if needed) and navigates to the balancing tab.

Each motor must be balanced separately, so Bob selects one of the motors to work with. He makes sure it's mechanically free to turn and indicates it to the configurator. The software spools the motor up to a certain RPM and vibrations are measured and a single figure (number) is shown to Bob. Then he sticks a piece of tape on the motor and reruns the test. He switches the tape around until the vibration figure he is receiving is minimal. He repeats the process for the rest of the aircraft's motors.

Next, he mounts the props and follows a similar procedure, this time turning each prop around (in relation to the motor bell) and/or adding tape to it. When he finds the prop orientation/tape quantity/tape location that causes the lowest vibration figure, he considers that motor/prop combo balanced, and repeats the process for the rest of the props.

Functional requirements:
FR1: Provide a way to start and stop each motor
FR2: Provide a way to set desired RPM and keep it stable for all tests. Hover RPM should be advised to be used. Using a percentage would be better than PWM signal.
FR3: Provide readings of vibrations, possibly including an "overall" figure that is easy for users to understand. This could be the average of the readings of the three axes, over the time of the measurement, minus spool-up and spool-down intervals, and maybe with some outlier removal. No graphs or multiple readings are required, just one number that contains most of the information and is easy for an inexperienced user to interpret.

FR4: Safety: Require the user to keep a button (space?) pressed while the test is running. If the button is depressed the test should immediately be cancelled and the motor instructed to stop.
FR5: Safety: if communication to the configurator is lost, stop all output to motors! (This probably requires changes to CF. Currently, if communication is lost, motors keep spinning the way they did. Dangerous!)

Non-functional requirements:
NFR1: Repeatedly performing tests should be easy and fast. Do not require user to fine-tune RPM using the mouse during every test.
NFR2: Taking the vibration reading and understanding if vibrations are more or less than the previous readings should be easy for an inexperienced user. (For later:) Maybe make it easy to compare vibrations between various runs, by saving previous rounds and instructing the user to either change the tape/prop or switch to another motor.

kgustine, 2016-01-22:
Glad to see this idea is already well formed, but sad to see there is no progress.  Is anyone doing anything with this at the moment?

closed 2017-02-25 by hydra