cleanflight/cleanflight-configurator #60: Consider publishing old versions of the Configurator in the Chrome Web Store

thenickdude, 2015-01-19:
If old (or even current) versions of the Configurator were published separately from the main "latest" version of the Configurator in the store, people could install these fixed-versions with one click and they'd never have to complain about it being updated from underneath them.

hydra, 2015-01-19:
It's an idea but not really a solution.  I think it would just create more confusion.

closed 2015-01-19 by hydra

nebbian, 2015-01-20:
Why was this closed?
What solution are you proposing?  The current situation causes frustration to _everyone_ who hasn't kludged their setup to stop auto updates.

thenickdude, 2015-01-20:
It's not really much friendlier than downloading old configurator releases
from the release page, and it introduces a new class of problems where
people flash newer firmware than their configurator supports and now have
to figure out how to upgrade their configurator (i.e. the reverse to the
problem that people are currently having).

Long term a lot of the backward compatibility issues will be fixed by
improvements to the MSP protocol for changing settings. Right now any
change in the settings that need to be saved pretty much breaks the
protocol (no forward or backward compatibility), which is dumb. But since
MSP protocol is used by a bunch of software that needs to interop with
Cleanflight, improving the protocol isn't easy either.

hydra, 2015-01-20:
@nebbian ^ what @thenickdude said. (thanks nick) :smile:

bbrummer, 2015-01-23:
Why not just have two in the store, Release and Beta?  Hell, even Chrome itself has a separate Chrome Beta in the app store for just this reason.

Update the stable/release on a standard schedule, say every 2 months (on the 1st of the month).  The Beta version can release at any pace, hourly if the devs want.  Everyone wins?