cleanflight/cleanflight-configurator #65: Flashing STM32F3 based targets in DFU mode from GUI

MJ666, 2015-01-23:
Looks to be the code already contain parts to enable DFU flashing? not sure what the actual state is and if this part of the code is already working. Actually the GUI is only allowing Hex files to be selected and will do some basic tests. Would be nice to be able to flash all targets easily via the GUI.

I have opened an related issue in CF already:
https://github.com/cleanflight/cleanflight/issues/412

MJ666, 2015-05-12:
@hydra this may be also get interesting for you with the SP Racing F3?

hydra, 2015-05-17:
SPRacing F3 uses CP2102, no DFU required.  Sparky requires DFU mode.

andymac4182, 2015-08-27:
@hydra Is there anything special other that shorting the boot pins to flash the SPRacingF3 if you disabled UART1 by mistake?

andymac4182, 2015-09-21:
I have been working through this and it seems mostly there. The only issue I have now is the getDevices (https://developer.chrome.com/apps/usb#method-getDevices) function doesn't return a device even through I can see it in DFU mode and flash it with other software. Any ideas or pointers?

tracernz, 2015-10-02:
Have you added the USB VID/PID to the manifest? It will only return devices you have permissions to access.

andymac4182, 2015-10-03:
They are added already in the existing Configurator. I confirmed they are correct. Been busy with work but going back to look at this soon.

tracernz, 2015-10-03:
They are listed as optional permissions though. You will need to make sure you request them somewhere. https://developer.chrome.com/apps/permissions#method-request

tracernz, 2015-10-03:
I had a look at this. You need to uncomment this line and then click the button on the landing page to request USB permissions: https://github.com/cleanflight/cleanflight-configurator/blob/master/tabs/landing.js#L13.

Then this line needs to go below line 356: https://github.com/cleanflight/cleanflight-configurator/blob/master/js/protocols/stm32usbdfu.js#L300

After that I successfully flashed my STM32F3 Discovery board using USB-DFU. I did note the code isn't really complete in that it doesn't handle some error conditions properly though.

andymac4182, 2015-10-03:
I will check this out again. I couldn't get it working on my version of Chrome.

tracernz, 2015-10-03:
I've submitted a PR which makes the minimal changes, please test that. :)

skaman82, 2016-06-16:
@MJ666 can we close this?

MJ666, 2016-06-16:
Sorry will close it.

PR #250 (2015-10-31): Enable flashing using USB-DFU protocol
For new boards like Tornado.
Fixes #65 .

Todo:
- [x] Auto-detect DFU mode after reboot
- [x] Implement local flash erase
- [ ] Detect chip ID/ flash page size for local erase
- [x] Make USB permissions compulsory
- [x] Add more error checking
closed 2016-06-16 by MJ666