cleanflight/cleanflight-configurator #72: Angle Flight Mode incorrectly activated

enly1, 2015-01-25:
I appear to have got cleanflight configurator into a mode where ANGLE wants to show as active even when the current channel receiver state is outside the activation range.

**NOTE that it only does it when no receiver signal is being received - regardless of value displayed on screen.**

See screenshot - AUX1 is setup for arm, angle and horizon modes all in same range.  Aux1 showing 1500 on screen and ARM and Horizon correctly disabled.  Angle however is showing green.

Removing it, adding it back in, saving and rebooting doesn't fix the problem.
Not sure if its a UI issue or actually the mode is really enabled.

[image]

[image]

20:52:06 -- Running - OS: Windows, Chrome: 40.0.2214.91, Configurator: 0.60.0
20:52:09 -- Serial port successfully opened with ID: 1
20:52:09 -- MultiWii API version received - 1.3
20:52:09 -- Flight controller info, identifier: CLFL, version: 1.6.0

-----------------------
Config Dump:

# dump

# version
Cleanflight/NAZE Jan 16 2015 / 12:40:10 (d72983e)
# dump master

# mixer
mixer QUADX

# feature
feature -RX_PPM
feature -VBAT
feature -INFLIGHT_ACC_CAL
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -FAILSAFE
feature -SONAR
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DISPLAY
feature -ONESHOT125
feature -BLACKBOX
feature VBAT
feature RX_SERIAL
feature FAILSAFE
feature SONAR

# map
map TAER1234

# led
led 0 2,2:ES:IA
led 1 2,1:E:WF
led 2 2,0:NE:IA
led 3 1,0:N:F
led 4 0,0:NW:IA
led 5 0,1:W:WF
led 6 0,2:SW:IA
led 7 1,2:S:WF
led 8 1,1:U:WF
led 9 1,1:U:WF
led 10 1,1:D:WF
led 11 1,1:D:WF
led 12 0,0::
led 13 0,0::
led 14 0,0::
led 15 0,0::
led 16 0,0::
led 17 0,0::
led 18 0,0::
led 19 0,0::
led 20 0,0::
led 21 0,0::
led 22 0,0::
led 23 0,0::
led 24 0,0::
led 25 0,0::
led 26 0,0::
led 27 0,0::
led 28 0,0::
led 29 0,0::
led 30 0,0::
led 31 0,0::

# color
color 0 0,0,0
color 1 0,255,255
color 2 0,0,255
color 3 30,0,255
color 4 60,0,255
color 5 90,0,255
color 6 120,0,255
color 7 150,0,255
color 8 180,0,255
color 9 210,0,255
color 10 240,0,255
color 11 270,0,255
color 12 300,0,255
color 13 330,0,255
color 14 0,0,0
color 15 0,0,0

set looptime = 3500
set emf_avoidance = 0
set mid_rc = 1500
set min_check = 1100
set max_check = 1900
set rssi_channel = 0
set rssi_scale = 30
set input_filtering_mode = 0
set min_throttle = 1150
set max_throttle = 1850
set min_command = 1000
set 3d_deadband_low = 1406
set 3d_deadband_high = 1514
set 3d_neutral = 1460
set 3d_deadband_throttle = 50
set motor_pwm_rate = 400
set servo_pwm_rate = 50
set retarded_arm = 0
set disarm_kill_switch = 1
set auto_disarm_delay = 5
set small_angle = 25
set flaps_speed = 0
set fixedwing_althold_dir = 1
set serial_port_1_scenario = 1
set serial_port_2_scenario = 3
set serial_port_3_scenario = 0
set serial_port_4_scenario = 0
set reboot_character = 82
set msp_baudrate = 115200
set cli_baudrate = 115200
set gps_baudrate = 115200
set gps_passthrough_baudrate = 115200
set gps_provider = 0
set gps_sbas_mode = 0
set gps_auto_config = 1
set gps_auto_baud = 0
set serialrx_provider = 2
set spektrum_sat_bind = 0
set telemetry_provider = 0
set telemetry_switch = 0
set telemetry_inversion = 0
set frsky_default_lattitude = 0.000
set frsky_default_longitude = 0.000
set frsky_coordinates_format = 0
set frsky_unit = 0
set battery_capacity = 0
set vbat_scale = 112
set vbat_max_cell_voltage = 43
set vbat_min_cell_voltage = 33
set vbat_warning_cell_voltage = 36
set current_meter_scale = 400
set current_meter_offset = 0
set multiwii_current_meter_output = 0
set align_gyro = 0
set align_acc = 0
set align_mag = 0
set align_board_roll = 0
set align_board_pitch = 0
set align_board_yaw = 0
set max_angle_inclination = 500
set gyro_lpf = 42
set moron_threshold = 32
set gyro_cmpf_factor = 600
set gyro_cmpfm_factor = 250
set yaw_control_direction = 1
set acc_hardware = 0
set mag_hardware = 0
set blackbox_rate_num = 1
set blackbox_rate_denom = 1

# dump profile

# profile
profile 0

# aux
aux 0 0 0 1700 2100
aux 1 1 0 1700 2100
aux 2 2 0 1700 2100
aux 3 12 2 1700 2100
aux 4 21 1 1700 2100
aux 5 0 0 900 900
aux 6 0 0 900 900
aux 7 0 0 900 900
aux 8 0 0 900 900
aux 9 0 0 900 900
aux 10 0 0 900 900
aux 11 0 0 900 900
aux 12 0 0 900 900
aux 13 0 0 900 900
aux 14 0 0 900 900
aux 15 0 0 900 900
aux 16 0 0 900 900
aux 17 0 0 900 900
aux 18 0 0 900 900
aux 19 0 0 900 900
aux 20 0 0 900 900
aux 21 0 0 900 900
aux 22 0 0 900 900
aux 23 0 0 900 900
aux 24 0 0 900 900
aux 25 0 0 900 900
aux 26 0 0 900 900
aux 27 0 0 900 900
aux 28 0 0 900 900
aux 29 0 0 900 900
aux 30 0 0 900 900
aux 31 0 0 900 900
aux 32 0 0 900 900
aux 33 0 0 900 900
aux 34 0 0 900 900
aux 35 0 0 900 900
aux 36 0 0 900 900
aux 37 0 0 900 900
aux 38 0 0 900 900
aux 39 0 0 900 900

# adjrange
adjrange 0 0 0 900 900 0 0
adjrange 1 0 0 900 900 0 0
adjrange 2 0 0 900 900 0 0
adjrange 3 0 0 900 900 0 0
adjrange 4 0 0 900 900 0 0
adjrange 5 0 0 900 900 0 0
adjrange 6 0 0 900 900 0 0
adjrange 7 0 0 900 900 0 0
adjrange 8 0 0 900 900 0 0
adjrange 9 0 0 900 900 0 0
adjrange 10 0 0 900 900 0 0
adjrange 11 0 0 900 900 0 0

set gps_pos_p = 15
set gps_pos_i = 0
set gps_pos_d = 0
set gps_posr_p = 34
set gps_posr_i = 14
set gps_posr_d = 53
set gps_nav_p = 25
set gps_nav_i = 33
set gps_nav_d = 83
set gps_wp_radius = 200
set nav_controls_heading = 1
set nav_speed_min = 100
set nav_speed_max = 300
set nav_slew_rate = 30
set alt_hold_deadband = 40
set alt_hold_fast_change = 1
set deadband = 0
set yaw_deadband = 0
set throttle_correction_value = 0
set throttle_correction_angle = 800
set yaw_direction = 1
set tri_unarmed_servo = 1
set default_rate_profile = 0
set failsafe_delay = 10
set failsafe_off_delay = 200
set failsafe_throttle = 1200
set failsafe_min_usec = 985
set failsafe_max_usec = 2115
set gimbal_flags = 1
set acc_lpf_factor = 4
set accxy_deadband = 40
set accz_deadband = 40
set accz_lpf_cutoff = 5.000
set acc_unarmedcal = 1
set acc_trim_pitch = 0
set acc_trim_roll = 0
set baro_tab_size = 21
set baro_noise_lpf = 0.600
set baro_cf_vel = 0.985
set baro_cf_alt = 0.965
set mag_declination = 0
set pid_controller = 0
set p_pitch = 40
set i_pitch = 30
set d_pitch = 23
set p_roll = 40
set i_roll = 30
set d_roll = 23
set p_yaw = 85
set i_yaw = 45
set d_yaw = 0
set p_pitchf = 2.500
set i_pitchf = 0.600
set d_pitchf = 0.060
set p_rollf = 2.500
set i_rollf = 0.600
set d_rollf = 0.060
set p_yawf = 8.000
set i_yawf = 0.500
set d_yawf = 0.050
set level_horizon = 3.000
set level_angle = 5.000
set sensitivity_horizon = 75
set p_alt = 50
set i_alt = 0
set d_alt = 0
set p_level = 90
set i_level = 10
set d_level = 100
set p_vel = 120
set i_vel = 45
set d_vel = 1

# dump rates

# rateprofile
rateprofile 0

set rc_rate = 90
set rc_expo = 65
set thr_mid = 50
set thr_expo = 0
set roll_pitch_rate = 0
set yaw_rate = 0
set tpa_rate = 0
set tpa_breakpoint = 1500

thenickdude, 2015-01-25:
You have feature FAILSAFE turned on, so it's probably going into failsafe mode since there's no receiver attached, which causes angle mode to be activated.

enly1, 2015-01-26:
Yup - thats it.  

Failsafe mode impacts the angle mode state (which unless you have it enabled on an aux, you never get to see). So just confusing in relation to the configuration modes and why it activates outside of the specified range.

This can be closed unless anyone wants to add an enhancement to make it clear that its been activated outside of range by failsafe.

hydra, 2015-01-27:
I have been thinking for a while that adding some indicators for various states to the UI would be good, including fialsafe and armed.  I will close this and open a new ticket.

hydra, 2015-01-27:
See #74

closed 2015-01-27 by hydra