cleanflight/cleanflight-configurator #74: Add state indicators

hydra, 2015-01-27:
Adding indicators for system status would be useful.

three important states are: calibrating, armed and failsafe.

hydra, 2015-05-17:
RX loss potentially available in addition to failsafe state now too.

Blackbox device capacity/state might be nice too - so you remember to download or erase it.

gaelj, 2016-04-10:
I have attempted to implement these features.
Please check and report any feedback on 
https://github.com/gaelj/cleanflight-configurator

gaelj, 2016-04-10:
Is RX_loss readily available in MSP or does it require to be implement in CF as well ?

skaman82, 2016-05-11:
Armed, failsafe, serial link and battery status are finaly added in https://github.com/cleanflight/cleanflight-configurator/pull/372

skaman82, 2016-06-16:
I think we can close this one

closed 2016-06-16 by skaman82