cleanflight/cleanflight-configurator #78: GPS Signal Strength Data Not Cleared

tracernz, 2015-01-28:
The GPS signal strength data is not cleared upon disconnect, reboot board (with GPS disconnected), and reconnect. Need to close the configurator and relaunch to clear it. I suspect this is happening as the MSP response to the sv info command only returns data for active satellites and the CLI musn't write anything beyond that to its internal array. That means if a satellite is lost it's old value will be retained which could be misleading.

tracernz, 2015-01-30:
You can see the issue at https://github.com/cleanflight/cleanflight-configurator/blob/master/tabs/gps.js#L50
The loop only writes up to the current number of satellites returned by MSP, leaving old data in the remainder on the cells. I tried rewriting the loop as below but it didn't seem to have any effect. Maybe Chrome wasn't reloading the extension properly, I'm not sure.
```javascript
        // 16 needs replaced with a var/constant
        for (var i = 0; i < 16; i++) {
            var row = e_ss_table.eq(i);
            if(i < GPS_DATA.chn.length) {
                $('td', row).eq(0).text(GPS_DATA.svid[i]);
                $('td', row).eq(1).text(GPS_DATA.quality[i]);
                $('td', row).eq(2).find('progress').val(GPS_DATA.cno[i]);
            } else {
                $('td', row).eq(0).text(0);
                $('td', row).eq(1).text(0);
                $('td', row).eq(2).find('progress').val(0);
            }
        }
```
If anybody can point out the error in my beginner javascript skills please do.

andymac4182, 2015-02-09:
@tracernz You are doing nothing wrong. The issue is further back up the chain than where you were looking. https://github.com/cleanflight/cleanflight-configurator/blob/master/js/msp.js#L458-L472 The data isn't being updated here. This is where you need to update the data to be 0 and the normal code will reset it out.

skaman82, 2016-06-22:
fixed in https://github.com/cleanflight/cleanflight-configurator/pull/399

closed 2016-06-22 by skaman82