cleanflight/cleanflight-configurator #93: Feature Request - Add power meter to Motor tab

SlyMike, 2015-02-08:
Just a thought... 

It would be great to see Volts, Amps and Watts on the motor tab with a graph if possible.
Maybe even MAX calculated values above the individual sliders and a RESET button to zero them again.

It makes for good eye candy, great for double checking that motor / prop combos wont kill ESC's mid flight and also for weeding out defective components. Plus there is plenty of empty space on the motors tab ;-) 

If the user could input cell count and C rating of the battery pack and link that to the above then this tab would be some next level shizzle.

I'd do it my self but all I know is that Java has something to do with coffee!

handsomejackuk, 2015-02-10:
would like to see this too  i wish i knew more about programming.. i did try virtual current meter the other day but will give it a few weeks. maybe it get sorted then. thanks all for the great software.. love it

hydra, 2015-05-17:
Yes, would be nice to show/graph voltage sag under load and current draw.  Good ideas!

Robomort, 2015-12-29:
Allowing / encouraging users to do power runs on thier motors while connected to the computer sounds very dangerous to me. This would add no functionality that could not be replicated via telemetry while the model is actually flying which is the important metric rather than a static test.
Cleanflight is already becoming bloatware loaded down with eye candy and irrelevant functionality let's at least keep it safe bloatware.

jtp755, 2015-12-29:
@Robomort Are you here just to bash CleanFlight and the Configurator?

What about people that do not have telemetry? This is not eye candy...it is actually useful. Dont work on your multi-rotor with the props on (as is stated in the warning that you have to enable) and problem solved.

closed 2017-02-25 by hydra