cleanflight/cleanflight-configurator #94: Migrate to AngularJS

andymac4182, 2015-02-09:
The idea behind a migration to AngularJS is that it would allow additional functionality to be added easily. It will enable the testing of the code with unit tests, dependency injection and simple internationalization. It will also allow for the easy porting through to Ionic Framework and Cordova with support for mobile devices to run the same logic with a different presentation layer and serial connection libraries. I have tested using Ionic and Cordova to connect to the Naze32 via USB and is all working correctly.

andymac4182, 2015-02-10:
@MichaelMcClenaghan and I are working on a initial UI migration for this. Once that is done we can plan out the move of backend stuff one step at a time.

hydra, 2015-02-14:
Looking forward to seeing this.  Please keep us posted, if you have a work in progress branch can you update this with a url to it?

andymac4182, 2015-02-16:
I have some pushed up work to start handling moving over the custom logic from the current configurator and separating out things like the serial layer we can replace with other layers in the future. 

https://github.com/andymac4182/cleanflight-configurator/tree/angularjs

VictorRogers, 2015-02-21:
Andy
I would be interested in getting involved with this aspect of the project if you guys are in need of any help. I'm still in school, so my free time is unfortunately somewhat limited. However, I have experience building a decent-sized Angular web application, and I have also been flying multi-rotors for about a year now. I've been wanting to get involved with an opensource FC for a while now, and this seems like a great spot to do it. Let me know!

andymac4182, 2015-02-23:
Sounds great @VictorRogers . I want to get the basic framework set out then we will have a lot of work to migrate each tab over.

enly1, 2015-02-23:
Hi - Not to be too cheeky, but when you get the framework up and running (Serial, FC Connection etc), before porting the tabs, can you make it available ?

I have another open source (To Be) project that I want to use the configurator for (non FC) and would save me unraveling all the tabs etc from the finished product.

MichaelMcClenaghan, 2015-02-23:
The nice thing about the way we are doing it is that all the serial stuff
will he AngularJS services, completely abstracted away from all of the UI
stuff.  So you should be able to just reuse those services (provided you
are going to use Angular as well.)

enly1, 2015-02-23:
Yeah, but will after large portions of the UI as well - so basically CF
Configurator, with different tabs, but talking MSP + Firmware updates etc.

Planning to re-use everything as unchanged as possible, but good to my
hands on early as it develops (not touched Angular before, so yet another
learning curve)

hydra, 2015-03-11:
@andymac4182 how is this coming along

Did you know angular 2 is out now?  How does this impact things?

pulsar256, 2015-03-13:
Angular 2 is currently in the alpha stage. API can (and probably will) change a lot before the release date (no ETA yet). It does also change a lot in the area of scopes and controllers. Expect the migration to be tough. On the other hand, I am pretty sure that the 1.x branch will be supported for quite a while after the release. After all it is widely deployed and adapted by the enterprise.

andymac4182, 2015-03-17:
I have been doing some reading about AngularJS 2. To me personally it is definitely the future and a lot faster than 1.x. After doing some work on doing the initial migration we really need to have a discussion first about how the new layout would work and what features to support. 

Sorry I have been a bit quiet recently. I had a ton of work dumped in my lap so have been sorting that out. Should be back on this end of this week start or start of next week.

MauroMombelli, 2015-04-26:
i like that you are using route, as i found out it keeps the code very clean, but seems like the development stopped.

andymac4182, 2015-05-23:
I have had to put this on pause due to a few large projects that came up. After doing some work with Angular and Typescript I can see that this would be the best solution. I think before doing a complete rewrite the best course of action would be to start abstracting the code from the UI as currently it is hard and basically requires a complete rewrite to move to angularjs.

MauroMombelli, 2015-05-23:
yes, the TABS folder has to be completly rewritten..

what i used to do in others project is to create a actory that save/load data structure form somewhere (json server in my case, the serial/usb in that case) This was just a sort of "DB" or "cache", being a factory it live with the application.
Basically almost all that live in "js" folder has to become a factory

then many partial with relative directive for every data structure; basically it take a js object and you tell how you want it to be represented. Those are the component of the TABS

and finally the "main" partials with relative controller, basically it has a couple of callback to take the data from the factory and put it in a variable, that then the partial loop and call the wanted directive for every element. Those are the actual TABS

andymac4182, 2015-05-25:
I think the ideal timing of this would be to start towards it along side https://github.com/cleanflight/cleanflight/issues/347

hillmanov, 2015-08-06:
Just wanted to see if there is more movement on this as well. I am an experienced Angular developer and would like to help out if possible.

pulsar256, 2015-08-06:
@hillmanov the migration is currently stalled. Can you join us on IRC? I think it might be easier that way to figure out if/how this can be done.

enly1, 2015-09-24:
Any update on this migration activity ?  Been hacking away with the main configurator for my other project, but before i actually invest too much time, was hoping to look at this opportunity and/or see if there is any help that I can offer etc.

closed 2017-02-25 by hydra