cleanflight/cleanflight-configurator #95: Edit LED Config GUI Tweak

Akfreak, 2015-02-09:
Show colors on numbered color buttons (configurator issue, moved from fc firmware issue tracker)

heathicus, 2015-02-26:
I just came here to make this same request.  Additionally, change the representations on the grid to show what color the "Orientation" light will be based on the orientation chosen rather than all green.

joefarmer, 2015-03-15:
Me three!

hydra, 2015-05-17:
@heathicus orientation colors are hard to repliate since they are only known by the fc and they depend on the mode.  the fc logic would have to be repeated in the UI.  If you want that please open a new issue.

This issue is for making the numbered color boxes show the colors used by the FC.  i.e. the colors that the cli `color` command shows.

Jonny360, 2015-06-06:
Would this also allow customizing colors for arming/indicator, etc? Because that would be great.

hydra, 2015-10-14:
@destos Wanna have a crack at this?

destos, 2015-10-17:
@hydra How open are you to me re-working the led configuration interface to use react.js? I can see many potential cleanup and simplification opportunities by using is instead of straight dom manipulation via jQuery.

skaman82, 2016-05-13:
@destos just give it a try. I would love to see with what you can come up.

skaman82, 2016-06-16:
Mostly taken care of in https://github.com/cleanflight/cleanflight-configurator/pull/390

skaman82, 2016-06-16:
This works now in the "special colors" section on the led tab

closed 2016-06-16 by skaman82