cleanflight/cleanflight-configurator #97: Sonar data not available at RC7 RC8

aikon96, 2015-02-09:
Hi there,

I'm using latest cleanflight & cleanflight configurator. I'm using a Naze32 rev.5 on a spider hex, so for the sonar I use RC7 / RC8 for trigger and echo. Before using this sonar in a flight, I wanted to see the sonar data in the GUI. But there is only "0" the whole time at sensor data, even when I trun the copter... So I didn't use the sonar mode yet. BTW: I tunred on feature sonar.

To check if the sonar "HC SR04" sensor is working, I put it on a raspberry pi and used a script to test it. And it works on my raspberry, also the distance is really accurate +/- some cm!!! So, there must be a bug in the Cleanflight / Cleanflight configurator!?

RC7/8 are also inline with 1kOhms, the sonar itself is powered on 5V, so the connection should be ok in my opinion.

Could it be possible that using sonar data at RC7/8 isn't working yet?
Can anyone comfirm this bug, using RC7/8?

Thanks in advance!

SlyMike, 2015-02-09:
No problem on my Naze32 with FW v1.7.1  Have you set "feature -RX_PARALLEL_PWM",  "feature RX_PPM" in CLI? 
Its defiantly not a configurator fault and its been working in the last few versions of the FW.

enly1, 2015-02-09:
Running 1.7.1 on naze32 with sonar on RC7/8 with 1k inline and no problems - results show cleanly on the sensor tab once sensor graph enabled ( and battery connected for BEC 5v supply to Sonar )

Running - OS: Windows, Chrome: 40.0.2214.111, Configurator: 0.61.0
Flight controller info, identifier: CLFL, version: 1.7.1
Board: AFNA, version: 512

Running RX_Serial (remember can't be using PWM)

aikon96, 2015-02-10:
Ok my bad, sorry.
I took the +5V from receiver port from NAZE32, so by connecting via USB, the sonar didn't work, but voltage was there...?! hm.... So after connecting Battery, it works now.

Thanks to enly1!

closed 2015-02-10 by aikon96