cleanflight/cleanflight-configurator #98: Set up wizard for new users

andymac4182, 2015-02-10:
It would be great to have a wizard for new users that helps them run through items such as failsafe, calibrate esc, Radio and other default things that people will need to change.

hydra, 2015-02-14:
agreed

andymac4182, 2015-02-16:
Here is the pages I think that should be in the wizard.
1. Mixer select
2. Board options
    * Oneshot
    * Board rotation (With diagram)
    * Motor spin at arm
3. ESC Calibration (Walking them step by step through it)
4. Failsafe Settings
5. Receiver channels (Maybe include 3d diagram to show as if in angle mode)
6. Mode set up
7. Tell to have fun :)

MasterZap, 2015-02-19:
Agree - look at OpenPilot which has a great wizard (and a very nice looking UI!)

I'd love to also assign the motors in the wizard, i.e. I just wire to whatever output, and then we figure out which motor is which in the wizard, and bam :)

closed 2017-02-25 by hydra