cleanflight/cleanflight-configurator #99: Move save button and add elements to adjustments tab

jistforfun, 2015-02-12:
Often the save button is at the bottom of a long tab, it would be more convenient to have the save button at the top of each page where the most important items are generally located.

The adjustments tab is a wonderful addition to CF!
Could we add a few more selectable elements to the drop down menu?

Level PID 
Alt PID 
VEL PID
Mag PID

wolfpackt99, 2015-03-19:
thought about this recently too. Perhaps the save button should be fixed to the bottom on the tab, outside of the scroll

hydra, 2015-05-17:
The features requested require firmware changes.  Please open a new issue in the firmware issue tracking requesting support for in-flight adjustment support for the things mentioned.  Once they're done we can work on the UI.  The changes are not hard, see:

https://github.com/cleanflight/cleanflight/commit/c174a52ebeaa7f5edbf5e3186db0b2ef02d24cea

and

https://github.com/cleanflight/cleanflight-configurator/commit/5f05e4c8cc4ce3f4e60b6ba1e8a1c5a0e2ade076

@wolfpackt99  or @jistforfun  Why not have a go?

skaman82, 2015-12-08:
Done this, will post a PR shortly when some other things are done

NightHawk32, 2015-12-08:
We will add a few things which require pushing the api-level, so we will collect them to avoid unnecessary increases of the api level.

skaman82, 2015-12-15:
done in PR #296

skaman82, 2016-01-07:
Merged in dev version, closing

closed 2016-01-07 by skaman82