cleanflight/cleanflight-configurator #104: Changing PID controller in PID tab does not load in proper values when appropriate

HFMan, 2015-02-13:
If one changes to pid_controller 2 (LuxFloat) via the gui, the float pid values are not loaded into the gui.  Clicking the Save button now saves the currently shown integer based pids to the float pids on the board, which are completely inappropriate.  

The converse is true as well- if I currently have been testing the LuxFloat controller and want to change to a non-floating controller, I can do so, but the PIDS are again not changed to the integer based pids.  Clicking Save now saves inappropriate values to the integer pids on the board.

This means the only correct way to change between floating and non-floating controller is to do it via CLI commands.  Nasty bug that may catch unwary users off guard.

hydra, 2015-02-14:
Yes, i think that changing the pid controller needs just needs to refresh the page.  I noticed this too when I added the dropdown.  Was waiting for someone to report it :smile: so well spotted. 

The issue is then that if the user has changed some values and then change the pid controller the changed values will be lost. So either we can handle that OR place a message next to the pid controller and profile dropdowns to at least tell the user it will happen.

HFMan, 2015-02-14:
I think it would be best to warn the user if they have changed any of the settings on that page and not saved yet (if you detect they have changed the pid_controller.)  Definitely easier to warn them with a static text warning near the drop-down, but you know how folks are!

hydra, 2015-03-10:
from IRC:

```
hydra: how about:
[01:20am] hydra: detect if user has changed PID controller OR pids
[01:20am] pulsar_: and mark it dirty
[01:20am] hydra: then disallow changes to other values from the ones changes
[01:20am] _3rdeyepro joined the chat room.
[01:20am] pulsar_: yes, i was about to elaborate on that option too.
[01:20am] pulsar_: nah, if the pid mask is not dirty, do not save it.
[01:20am] hydra: then when saving force a refresh of both pid controller and pids
[01:20am] pulsar_: or mark it read only on pid change. yes.
[01:20am] hydra: and only save the ones that have been changed
[01:21am] pulsar_: perhaps a better option.
[01:21am] pulsar_: agree
[01:21am] hydra: and, as you say, only save the things that are changed
[01:21am] pulsar_: pid change -> if pid values were touched let the user confirm to loose the edited values
[01:21am] hydra: yes
[01:21am] pulsar_: otherwise gray out the pid values after a change and refresh on save.
[01:22am] hydra: yes
[01:22am] pulsar_: and enable ediiting.
[01:22am] pulsar_: this will work.
[01:22am] hydra: agree
[01:22am] hydra: ok, gimme PR 
[01:22am] pulsar_: tomorrow.
[01:22am] hydra: i look forwards to it
[01:22am] hydra: yes, late here too
[01:22am] hydra: i’m off to bed
[01:22am] pulsar_: yeah, good idea.
[01:22am] olexs left the chat room. (Quit: Leaving)
[01:22am] pulsar_: nn
[01:22am] Tobad_: another thing..  PIDs..  LEVEL P is angle and I is horizon..  changing values wont change level_angle or level_horizon in cli
[01:23am] hydra: i will make the issue as in progress
[01:23am] pulsar_: do that.
```

hydra, 2015-03-11:
This is now fixed in latest development configurator, please test and re-open if there's still an issue.

closed 2015-03-11 by hydra