cleanflight/cleanflight-configurator #105: Can't Visualize 3D view in the setup tab

Klakinoumi, 2015-02-15:
On my Macbook running 10.7.5 and equipped with a GMA 950, since cleanflight 0.60, I'm not able to visualize the 3D view of my quad in the setup tab. Heading info are working fine.

Packet errors are also appearing when i'm in this tab ( per sec), and only when I'm in this tab. Everywhere else it's fine. But not in the setup tab.

Quad is running fine. This is the only issue I have.

Let me know if you need more info on this.

thenickdude, 2015-02-15:
What mixer is your craft using, is it a custom mix? I think there is already a request to fix the missing model for the custom mix.

Klakinoumi, 2015-02-15:
I d'on't remember setting something particular on this. Where can I check this ?

stronnag, 2015-02-15:
I also no longer have 3D on my chromebook (Arch Linux), which is not OpenGL capable. I suspect the difference is that the old fallback.json worked without GL, and new the shiny AKFreak variant missed the whole point of ***fallback*** (the clue's in the name). 

Replacing the current (large) fallbacl.json by the previous version restores functionality on weak graphics cards. Please consider reverting.

stronnag, 2015-02-15:
I would also hazard a guess that the effort of a weak graphics card trying to process the OpenGL stuff in software causes the packet loss (as I also saw today) for a  quad_x. I'd raise a reversion PR if I though I  had a licence to redistribute the old stuff.
In hindsight, explains the in-field configurator issues experienced today.

stronnag, 2015-02-15:
i.e. see tabs/setup:209-214 for why fallback is much more than just for missing models. And my other netbook is a c. 2009 Asus model with equally weak GPU (so chromium by default disable GPU rendering) --> no model in the UI, regardless of frame type.

thenickdude, 2015-02-15:
The fallback model doesn't load properly on either OpenGL or the software renderer, it looks like the export options from Blender aren't correct (wrong file format and the scale is too large to fit on the screen).

hydra, 2015-03-11:
This should be fixed now, see referenced issues and PR's.  If not please re-open.

thenickdude, 2015-03-12:
It's fixed on OpenGL but not completely fixed on software rendering (though the error rate should be cured by @hydra's check to see that the model is loaded before it is attempted to be rotated). On software rendering it currently looks like:

[image]

hydra, 2015-03-12:
@thenickdude ahh ok, thanks for the update.

Akfreak, 2015-03-12:
Ill fix it.. I was hoping this would not be a problem. I see some folks insist on running machine with not much capability. I dont get it, but Ill fix it none the less.

thenickdude, 2015-03-12:
Yeah we fixed the model file which used to be using the wrong settings, so now it works on OpenGL. However on software rendering the triangles that make up the face of the cube sometimes draw in front of the text due to the crappy way that software rendering sorts triangles from front to back while drawing. 

The fallback model will basically have to be a textured cube in order to solve that problem.

Akfreak, 2015-03-29:
Fixed

closed 2015-05-17 by hydra

timdonovanuk, 2016-04-01:
Quick question...is the display of the 3D model in any way linked to the FC (other than using acc for animating the movement of it). I crashed my quad into water, and in CF config I now get a 3D cube instead of a 3D quad model. The cube does move appropriately when I switch on the accelerometer..Odd!

stronnag, 2016-04-01:
Whether you get a box or a model depends on two things:
* If the browser / hardware supports WebGL
* You're using a standard (vice custom) mixer

So it's less likely to be a FC setting, unless your aquatic excursion was corrupted the settings.

timdonovanuk, 2016-04-01:
Very strange then. It's still set to the default X quad and a CLI dump confirms this. I promise I didn't drop my laptop into the water too :) Cheers for the quick response

timdonovanuk, 2016-04-01:
I guess the FAA finally got their way and have permanently turned my quad into a stick of butter, even after a reflash.

[image]

skaman82, 2016-06-16:
@Akfreak Do you have some nicer alternatives for the custom mixer model?