cleanflight/cleanflight-configurator #123: Artificial Horizon in Cleanflight configurator is wrong -  IE pitch & roll are reversed

Rob2160, 2015-03-01:
Just noticed this.

When you pitch nose down, your artificial horizon should rise - when you roll left the left side of the horizon should rise.

Currently the display is opposite to how a real artificial horizon works.

See my video here.

https://www.youtube.com/watch?v=fhwFLmB2H0M

This is how it should display.

https://www.youtube.com/watch?v=dsCt88b5lwI

https://www.youtube.com/watch?v=PUmJOFrIV-c

tricopterY, 2015-03-01:
In the meantime can you download the configurator from my fork, and copy it to your cleanflight-configurator folder.
https://github.com/tricopterY/cleanflight-configurator

I will create a pull request for the fix.

hydra, 2015-03-11:
fixed

closed 2015-03-11 by hydra