cleanflight/cleanflight-configurator #133: Pid tuning: reset to defaults/ values per Pid controller

RandGH, 2015-03-14:
It would be nice if there was a reset to default values button in the gui to revert any ill modifications being done.

Additionally it would be great if PIDs could be saved *per* Pid Controller as those need very different values and other PidC's values might actually be harmfull.

peisel, 2015-05-12:
This would really helpful!

hydra, 2015-05-18:
The FC would need to expose PID defaults to the GUI.  Please open an issue in the firmware issue tracker and link to this.

hydra, 2015-05-18:
@RandGH please keep issues focussed.  One issue per request.

This issue is for allowing pids to be reset to the firmware defaults.  With regards to the other topic you mentioned.  No.  Though I think changing pid controller should probably reset all pids to their pid controller defaults.  Please open a new issue to discuss this second topic further and provide a link to this comment in it.

jrauch67, 2015-07-28:
Hi, have you created the issue to expose PID defaults in the firmware issue tracker ?

skaman82, 2016-06-16:
already done

closed 2016-06-16 by skaman82