cleanflight/cleanflight-configurator #137: PID Tuning tab - disable irrelevant fields per PID Controller / Sensors?

pfriedel, 2015-03-16:
It might be helpful for new people to disable the PID settings for hardware that doesn't use them anyway - i.e. on a Gyro/Accel board, having PID settings available for Pos/PosR/NavR only adds to a wall of numbers that won't have any affect on flight performance.

Broadly speaking, I'm thinking that a Naze32 Acro should only show ROLL, PITCH, YAW, and LEVEL, I'd think?  ALT requiring a baro, VEL, Pos, PosR, NavR requiring GPS.

And similarly, if there are any PID controllers that don't use a term (PI only, PD only, etc), disabling the term that isn't doing anything anyway would help people understand that tuning that dial won't affect anything.

ummagawd, 2015-03-26:
good idea. I'd also add to remove any other variables that are unaffected..... such as TPA for certain PID controllers.

hydra, 2015-05-18:
Agreed, this will certainly help new users.

wwkayak, 2015-10-29:
I actually never really noticed them them until @DigitalEntity started his fork, now I'm back looking at exactly what they are... so ya, not needed.

skaman82, 2015-12-04:
This is done already in 1.0

closed 2016-06-16 by skaman82