cleanflight/cleanflight-configurator #138: Select UART on version 1.8

Vincent81, 2015-03-18:
Hello,
On 1.8 selecting UART is bad and does not work.
flash the 1.7 puts everything well.
[image]

rocky77, 2015-03-18:
Please, read the releases notes.  You need CF-configurator v0.63!

--- Message initial ---

https://github.com/cleanflight/cleanflight-configurator/issues/138

Vincent81, 2015-03-18:
thank you for your answer.
I had searched an update, but I found nothing.
I even tried the zip locally ..
Can you tell me where to find this version?

rocky77, 2015-03-18:
Here : https://github.com/cleanflight/cleanflight-configurator/tree/baseflight-configurator-development

--- Message initial ---

https://github.com/cleanflight/cleanflight-configurator/issues/138#issuecomment-83005663

Vincent81, 2015-03-18:
Ok I just found.
Yesterday I had to download the correct zip.
thank you.

closed 2015-03-18 by Vincent81