cleanflight/cleanflight-configurator #142: Multiple aux channels to enable mode not working

Bmooreky, 2015-03-20:
When I choose two channels (Aux1 and Aux2) to enable Horizon mode (or any other mode), the mode will set when only one is in range.

hydra, 2015-03-30:
Some modes override other modes by design.  e.g. Horizon / Angle.  If it's a bug please provide a screenshot and open an issue for the firmware in the firmware issue tracker.

closed 2015-03-30 by hydra

thenickdude, 2015-03-30:
I think Bmooreky is saying that they want to only enable Horizon mode when *both* AUX1 and AUX2 are within a certain range. However, I think the system is built so that the mode is enabled when *any* of the channels are in the range.

Bmooreky, 2015-03-31:
thenickdude is correct, I had seen a youtube from someone that suggested that multiple aux channels could be combined to select a mode. If that's not the case, and it's FAD (functioning as designed), that's all I needed to know. Thanks for your feedback and the collective work on a great software and firmware.