cleanflight/cleanflight-configurator #143: Failsafe settings not present in backup

skyflyer, 2015-03-21:
When I backup the settings using Cleanflight Chrome app, only one failsafe setting is written to backup json file -- `failsafe_throttle`. Other values are missing. They are visible in the CLI dump, however.

hydra, 2015-03-21:
Currently he configurator can only back up settings which are exposed via MSP.  The other settings you mention are not suppoted.

tricopterY, 2015-04-16:
@hydra - I am working on this enhancement with introducing new MSP codes:
```C
#define MSP_FAILSAFE_CONFIG             75 //out message         Returns FC Fail Safe settings
#define MSP_SET_FAILSAFE_CONFIG         76 //in message          Sets FC Fail Safe settings
```
Just waiting for the 1.9.0 fix related to cleanflight/cleanflight/#774

hydra, 2015-04-16:
don't work on this until the new failsafe code is merged in to 1.9.0 please otherwise it will need re-work.

tricopterY, 2015-04-16:
Thanks for the heads-up, I will wait.

hydra, 2015-05-17:
Should be fixed, please re-open if not.

closed 2015-05-17 by hydra