cleanflight/cleanflight-configurator #147: Combine PID controller 1 and 3

ummagawd, 2015-03-26:
Hi all - my apologies if this is the wrong place for feature requests, first time github user here.

Anyways... I think that if the roll/pitch code of PID controller 1 can be somehow combined with the yaw code of PID controller 3... we would have the ultimate PID algo.

PC1 has terrific roll/pitch characteristics, and easy to tune. Very very tolerant to vibrations, and I can tune this to the point of zero shakes/oscillations (to the point that you wouldn't notice it in HD video feedback at least). But my complaint is that yaw authority diminishes with roll/pitch input.

PC3 on the other hand, has terrific yaw characteristics, but is overpowering roll/pitch input. Its very flyable however and my favorite for all types of flying, but my complaint is that roll/pitch is sensitive to vibrations and hard acro moves. I along with others have played with the tune for a while, and can't seem to tune out the shakes/oscillations.

Possible to get the best of both worlds?

tricopterY, 2015-03-27:
I use PID Controller 3, and I was having the same challenges getting rid of oscillations and wobbles. The idea is to use TPA related params like tpa_breakpoint and tpa (throttle PID attenuation) to help with the tuning. Use the latest Configurator i.e 0.63.0 and the pid tuning tab lets you set those values.
You may have to increase your I values for Roll/Pitch if your craft wobbles whilst descending in its own prop wash. I have managed to get that locked in feeling on a Naze32 acro (almost right on my mini quad). If your craft oscillates on throttle punches, decrease your P values for Roll/Pitch and tweak the tpa say to 0.50 and tpa_breakpoint to 1450 (if you hover just under 50% throttle).

stinger103, 2015-03-27:
Ok thank you for the info👍

Andrew del Castilho

hydra, 2015-03-30:
This ticket has been posted in the wrong issue tracker.  This tracker is for the configurator.

closed 2015-03-30 by hydra