cleanflight/cleanflight-configurator #149: Custom mixer editor

toasthall, 2015-03-28:
I have been thinking about an editor for custom mixers. What I envisage is a graphical interface with a graph that adds the motor location by entering the pitch and roll distances from the centre of the frame (or FC location). Calculation would be 1 divided by the value of the largest axis (ignoring the sign) then multiplying that figure by the distance for the particular motor.

I think this would simplify mixes for custom frames or frames like spiderquads, v-tails or a-tails (can't figure out yet how to calculate values to take account of motor angles)

hydra, 2015-03-30:
yes it would be very useful.  Fancy having a go at it?

toasthall, 2015-03-30:
Yes I do, the problem is the last time I did anything HTML was in the 90's. Might still give it a go though.

toasthall, 2015-04-01:
Hi Dominic,
Not sure how this all works. Is there any info or templates that might be useful for me in regards to attempting this?

cheers,

Dale

tricopterY, 2015-04-01:
I would be interested in this too. The calculation should first attempt, finding the Centre of Thrust, using the typed in values relative to the Roll/Pitch axis. Then create values for motorMixer table, taking into account Yaw rotation signs. 
Also, template should suggest best mounting position for FC i.e CoT closest to CoM.
@toasthall - you can use Configurator Tabs, as a template, led_strip.html and led_strip.js, should give you a heads-up.

toasthall, 2015-04-02:
Thanks @tricopterY 
I am still trying to wrap my head around the maths behind it all. I think to begin with, to make it simple for me, I will base the measurements off of the center of the FC board which i think should give a reasonable indication of the board's location in regards to the motors, that way people can play with board placement as well.

tricopterY, 2015-04-02:
Just a bit of hint - it's best to calculate the Center of Thrust against two imaginary axes eg Y-roll axis and X-pitch axis, at an offset relative to the center of the FC board. The maths would take care of the calculated measurements, as you would be measuring a certain distance from Y and X datums.
Y
|     M1       M2
|            +
|___________  X

The plus sign represents the calculated CoT relative to Y and X. 
Asterisk represents multiplication.
CTx = (T1 * M1X1 + T2 * M2X2) / T1+T2
In your calculation you would use Unity (one) for values T1 and T2. 
M1X1 and M2X2 are distances of motors M1 and M2 along X axis from Y axis, hence CTx. 
You would repeat the same for X axis i.e CTy = (T1 * M1Y1 + T2 * M2Y2) / T1+T2

toasthall, 2015-05-27:
Hi Guys,

Turns out I have no idea what I am doing with regards to developing this, if anyone else is interested in giving it a go that would be great.

Dale

skaman82, 2015-12-04:
will look into this in the next weeks

Rob2160, 2016-01-04:
Here you go.  Does V-Tails also. 

https://dl.dropboxusercontent.com/u/62849459/NAZE%20MIX%20Calculator%20V5.xlsx

skaman82, 2016-01-07:
Working on this with @NightHawk32. The progress is posted here: https://github.com/skaman82/cleanflight-configurator/issues/54
Starting with a Quad X Mixer interface.

skaman82, 2016-06-16:
@NightHawk32 any news on this?

closed 2017-02-25 by hydra