cleanflight/cleanflight-configurator #153: Magnetic declination : incoherent description between CLFL and GUI

Pierre-A, 2015-04-02:
The magnetic declination coding is described in the *config_profile.h* file of Cleanflight :  [sign]dddmn (minutes next to degrees) and presented that way in CLI.
Example is given for japan : -6°37' which becomes  -637 for the CLFL int16-t *mag_declination*.
In the GUI (tab *Configuration*), this is presented as deg (with mn as decimals) which is not consistent and source of questionning and error.

tricopterY, 2015-04-03:
Well, that is very interesting - the mag declination should be calculated eg for Japan:
-6 + 37/60 which is -6.616
I guess, it's a lame way of coding, do not bother with the minutes. In reality the above example is misrepresented as -6.4 vs -6.6. Is it worth fixing? May be a definite Yes, just for brevity.

skaman82, 2016-05-13:
Done in: https://github.com/cleanflight/cleanflight-configurator/pull/369

closed 2016-05-13 by skaman82