cleanflight/cleanflight-configurator #158: Configuring Receivers should be easier, maybe with a drop down of popular configurations.

NullVoxPopuli, 2015-04-10:
It should be possible to select "popular" receivers, and have the configurator set whatever parameters it needs to, so that there is less command line / apparent not saving due to feature conflicts.
Maybe there could even be pictures of each receiver in case someone somehow didn't know what receiver they had?

Obviously, an other option would be needed to have control over all the RX_SERIAL / whatever other parameters are involved in processing RX signals.

hydra, 2015-05-17:
Agreed, e.g. for SBus it's a bit of a faff, port selection first, then feature, then type.

hydra, 2015-05-17:
Perhaps a recevier setup wizard is needed.

Propitiation, 2016-01-30:
agreed, yes please make it easier/possible to use my futaba sbus receiver. I have yet to fly with cleanflight because of this. Baseflight won't let me use my shiny new leds...

closed 2017-02-25 by hydra