cleanflight/cleanflight-configurator #164: Suggestion: Selectable 3D Model in GUI

Arakon, 2015-04-25:
When using a custom mixer for i.e. a quad (which can make sense with the H-Quads for example), it would be nice if it was possible to select a Quad model instead of the generic box. A simple right-click selection menu maybe.
Also, a note on the 3D Model screen might be useful, saying that the propeller directions shown do not reflect the actual propeller directions needed. The model (at least for the quad) has reversed propellers (upside down) and propellers for the wrong directions (i.e. CW instead of CCW).. just to avoid confusing newbies.

hydra, 2015-05-17:
This is problematic if you have more than one quad with a custom mixer.

skaman82, 2015-12-04:
@hydra why don't abandon the ugly 3d box and just show the quad model even for custom mixer models? That is so much nicer an the directions are pretty clear.

t413, 2016-02-11:
+1, after seeing such a nice 3d model it's hard to stare at the box. Perhaps a dropdown or menu-item to select which existing 3d models are included already or add a custom one.

closed 2016-06-16 by skaman82