cleanflight/cleanflight-configurator #175: GPL Licence vs Apple store

blezalex, 2015-05-05:
I made a port of this configurator to IOS, i'm using ESP8266 dongle for wifi connectivity. 

I want to publish this app as a free app to apple store (no ads), yet apple policy bans GPL license in the store. 

Is it possible to re-license this app under less restrictive license? 

I think a lot of people will benefit from getting this configurator on mobile devices. I'm using Apache Cordova to make this run on IOS, it also allows to run the app on Android on Windows phone. 

I have it running on my phone today I want to share it with others :)

thenickdude, 2015-05-05:
Doubt it, because the main copyright holder of the code (the Baseflight Configurator upstream) is now hostile to open source and has closed the upstream project. Every copyright holder has to agree in order to add a new license option.

blezalex, 2015-05-05:
I wonder if Baseflight Configurator would like to have their configurator on mobile phones as well.
In the other hand, if they dont, having cleanflight on mobile and having consistent experience across desktop and mobile could be a big advantage for cleanflight. 

How much code still remained from baseflight in cleanflight? I wonder if those parts could be replaced. If feels very bad to be a hostage of baseflight restrictive license forever. 

As a different alternative i could try converting this app into a website hosted on ESP8266 wifi dongle, yet that would require a lot of compression and removal of many dependencies as code has to fit 200kb in compressed form. Updating such site is also much harder..

anvaka, 2015-05-06:
+1 to @blezalex request.

hydra, 2015-05-17:
@blezalex Publish the source to your one and perhaps we can use some of it, in the mean time other users would be able to compile and run it too without needing the app store.

grahamwho, 2015-05-19:
Yeah this would be awesome, great work

andymac4182, 2015-05-25:
As part of the rewrite of the communication stack https://github.com/cleanflight/cleanflight/issues/347 we could look at doing a complete rewrite and look at the licencing then. https://github.com/cleanflight/cleanflight-configurator/issues/94. Really depends how much time everyone has.

AntonieB, 2015-06-08:
I'm not really 'contributing' anything to this discusion but I guess it would be awsome if there was some kind of IOS application like 'cleanflight configurator'.

blezalex, 2015-06-13:
https://github.com/blezalex/MobileCFConfigurator - sources for mobile app 
https://github.com/blezalex/Wifi-Cleanflight-Configurator - sources for ESP8266 firmware enabling Wifi for naze board.

datamafia, 2015-07-22:
The alternative I am researching is getting a tablet with an Intel (x86) chip tablet such as Atom that also has a USB port. From there boot to Ubuntu and Chrome+Cleanflight. 

I don't want Bluetooth/wifi/etc config. I am fine using a USB cable in the field. I would love to find a few others who have had success/failure/other.

Even if CF was on iOS, the lightning plug is far from friendly, I don't want to add more gear and little circuit boards to my build, BT is not helpful. 

There is an Android port of the Silab driver, but I am not familiar w/any Android CF config app.

I encourage a rewrite, the upstream person should be cut out of the equation.

AustinSaintAubin, 2015-08-03:
Chromebook's work well. Have tested cleanflight with Samsung Chromebook with no issues. Can even easily save co figs to Google Drive.

proggod, 2016-02-27:
Could this be made to work with other boards, or is naze32 have something special to allow the wifi dongle to work?

blezalex, 2016-02-28:
I'm not a licensing expert, but it might be possible to create shell app for this and publish in appstore, then shell would download vanilla cleanflight from github.. Not sure if that is ok with licensing though

tracernz, 2016-02-28:
Heh, the vast majority of the code remains from baseflight configurator so this isn't likely to happen unless you can work around the app store issue.

datamafia, 2016-02-28:
I got a Galaxy S3 with EZGUI for field adjustments. Works great. The USB OTG can suck. Contact me direct if you want some tips. @tracernz is right, the iOS licensing is the deal killer, why I went droid.

closed 2017-02-25 by hydra