cleanflight/cleanflight-configurator #179: Demo mode or even Remote Configurator Help

MassiveOverkill, 2015-05-07:
I've been helping customers a lot lately with configuration and as I don't have all the screens memorized I'll walk them through the screens as I go through them myself. Unfortunately I have to connect my board in order to do this, which isn't always convenient especially when I'm traveling.  It would be nice to have a demo mode where I can go through the screens without having to hook up an FCB or even if possible Remote CF Configurator, sort of like remote desktop.  Now I can see right off the bat motor tab testing needing failsafes if this were even possible\feasible to implement.

tricopterY, 2015-05-07:
A poor man's solution would be to take ScreenShots/PrintScreen of each tab and save them as an image (jpeg extension) with correct file name eg Setup.jpg, PID Tuning.jpg etc. You can then use an image viewer to preview the images. A remote desktop is out of the question, but a Demo is feasible. Best solution would be to use a screen recording software. I am sure there are some on the net that you can download for free.

MassiveOverkill, 2015-05-07:
This is all fine if CF GUI remains static.  As you can see from this post:  http://www.rcgroups.com/forums/showpost.php?p=29485462&postcount=2  trying to keep up with accurate screenshots is tedious and futile.  I'm aware of other free remote desktop solutions.  The problem with those is time wasted trying to talk the customer how to install\configure them vs an integrated solution.  If it's not possible, then it's not possible, but it doesn't hurt to ask. I'll be happy with Demo Mode ;)

tricopterY, 2015-05-07:
Sorry, may be I misunderstood the issue. Remote desktop is there via the OS, if your customer allows you access. But to actually do it via CF is out of the question.

MassiveOverkill, 2015-05-07:
Are you speaking from a technical standpoint, a security one, or both?  'like remote desktop' was meant to imply 'remote connectivity' and ONLY to the CF app, not to imply complete PC control.

hydra, 2015-05-07:
There could be an FC simulator that responds to MSP. This would help with automated testing too.

tricopterY, 2015-05-07:
@hydra - I may be interested in writing a simulator as per your insight. Do you think there will be real value/take-up, for the amount of effort involved?

hydra, 2015-05-18:
Please raise one feature per issue.  There are two topics:
1) demo mode.
2) remote connection.

Let's keep this issue open to discuss 1.  Issue 2. can be solved if the user exposes their serial port over the internet, perhaps more simply solved by using a window or screen sharing program; this is out of scope of the configurator UI and likely not even technically possible within the confines of a chome app which is already sandboxed.

MassiveOverkill, 2015-05-18:
I've edited the OP for #1 functionality only.

heathicus, 2015-07-10:
I'd like to request this feature as well.  A "demo" or "offline" mode to access all of the tabs of the configurator without having to connect to the flight controller would be very beneficial and handy.

closed 2017-02-25 by hydra