cleanflight/cleanflight-configurator #183: TPA implementation to PID1 and PID2

borisbstyle, 2015-05-08:
PID1 and PID2 are missing TPA feature, which can be implemented very easily from what I see in the code. It happens that these seem to be very popular lately.

For example for PID1

This line:
PTerm = (RateError * pidProfile->P8[axis]) >> 7;

Could be replaced by:
PTerm = (RateError * pidProfile->DynP8[axis]) >> 7;

etc. etc.

Is there anybody who sees issues with this?

thenickdude, 2015-05-08:
The dynPIDs do include the effect of TPA but from memory they're also the mechanism by which the rotation rates are implemented (reducing pid response at edge of stick). So those values can't really be used without completely rebuilding the PID controller.

borisbstyle, 2015-05-08:
@thenickdude What about writing a different function for TPA, which only reduces PID's and does nothing with coupling?

borisbstyle, 2015-05-08:
This part of code is very ugly to me in all multiwii versions. Why the hell should you use this copling scaling?

// FIXME axis indexes into pids.  use something like lookupPidIndex(rc_alias_e alias) to reduce coupling.
        dynP8[axis] = (uint16_t)currentProfile->pidProfile.P8[axis] * prop1 / 100;
        dynI8[axis] = (uint16_t)currentProfile->pidProfile.I8[axis] * prop1 / 100;
        dynD8[axis] = (uint16_t)currentProfile->pidProfile.D8[axis] * prop1 / 100;

borisbstyle, 2015-05-08:
Ok I see that we could re-use the prop2 value somehow to be applied to the current PID's as reducer without adding prop1 to it. That could help to only reduce the P, I or D without adding coupling.

borisbstyle, 2015-05-08:
@thenickdude 

What about this. New parameter reduced P, I and D. If you use redP8 in PID controller 1 instead of P8 you will have only reduced the P factor without coupling.

        // FIXME axis indexes into pids.  use something like lookupPidIndex(rc_alias_e alias) to reduce coupling.
        dynP8[axis] = (uint16_t)currentProfile->pidProfile.P8[axis] * prop1 / 100;
        dynI8[axis] = (uint16_t)currentProfile->pidProfile.I8[axis] * prop1 / 100;
        dynD8[axis] = (uint16_t)currentProfile->pidProfile.D8[axis] * prop1 / 100;
        
        // non coupled PID reduction used in PID controller 1
        redP8[axis] = (uint16_t)currentProfile->pidProfile.P8[axis] * prop2 / 100;
        redI8[axis] = (uint16_t)currentProfile->pidProfile.I8[axis] * prop2 / 100;
        redD8[axis] = (uint16_t)currentProfile->pidProfile.D8[axis] * prop2 / 100;

borisbstyle, 2015-05-08:
Feature is added.
https://github.com/cleanflight/cleanflight/pull/862

I notice that I opened this issue for cleanflight configurator rather than cleanflight. Sorry for that :(

closed 2015-05-08 by borisbstyle