cleanflight/cleanflight-configurator #185: Configurator rejects development version of firmware

zdar, 2015-05-09:
HI,
I'm not sure if it is a bug or feature but configurator rejects firmware reporting higher version of MultiWii API then it knows.  E.g. I have current development firmware which has bumped API version recently and stock Configurator.

If it is feature than I suggest that message has to be changned in a way that it reports minimum and _maximum_ API version required.

2015-05-09 @ 15:18:58 -- Serial port successfully closed
2015-05-09 @ 15:19:01 -- Serial port successfully opened with ID: 47
2015-05-09 @ 15:19:04 -- MultiWii API version received - 1.10
2015-05-09 @ 15:19:04 -- This firmware version is not supported. Please upgrade to firmware that supports api version 1.2 or higher. Use CLI for backup before flashing. CLI backup/restore procedure is in the documention.

danielgpalmer, 2015-05-10:
I think this is due to the configurator parsing the version as a float instead of as two integers (major and minor) separated with a dot. So 1.10 is smaller that 1.2.

tricopterY, 2015-05-10:
Grab the latest version of configurator - it uses a semantic versioning library to cater for versions like 1.9.0 and 1.10.0 etc.

hydra, 2015-05-17:
Already fixed on v0.63.2

closed 2015-05-17 by hydra