cleanflight/cleanflight-configurator #189: Configurator should be able to Back up (and restore) the uav by a full CLI dump

MasterZap, 2015-05-18:
One of the most annoying parts of the Configurator is that saving to file doesn't save any CLI settings. This is kinda silly. The save routine should really do what everyone is told to do manually (a CLI dump) and the "Restore" routine should do the opposite.

Or, ideally, add on the serial protocol a method to return every CLI variable in sequence of name/value pairs, as well as setting them (which amounts to the same thing without having to "actually" go to CLI)

/Z

tricopterY, 2015-05-19:
For the time being a workaround is to do a dump, then select every line in the CLI dump screen and save it in a text file. To restore, you can select the lot from your text file and paste into CLI command field and hit enter key. Then type save and hit enter key again.

MasterZap, 2015-05-19:
Of course you can do it manually, but the Configurator should be about ease-of-use. 

Actually ... I wonder if the CLI makes sense at all for anything but us computer nerds. Since (nearly) the only thing the CLI can do, is to get/set variables, I wonder if it shouldn't be hidden away under a general get/set variable paradigm, that could easily be added to MSP.

That way, each PID controller could, for example, only expose the variable *it* needed, with the defaults *it* wanted under the names that suited *it*. None of this nonsense that "The D term in Level actually means percentage of horizon mode in PID controller blah-di-blah".

But... for starters.... just a built in CLI-dump / CLI-restore function would go a long way :)

hydra, 2015-05-19:
yes, the method of backup/restore is bad since it only allows the configurator to backup settings that are exposed via MSP.

I have talked in the past about using a different storage system in the firmware for configuration items but this requires some extensive rework.    See https://github.com/cleanflight/cleanflight/issues/347

the cli *will not* be used for backup/restore purposes.  that is just like putting a bandage on the code and not fixing the real problem.

This is blocked until such a system is present in the firmware.

MasterZap, 2015-05-19:
That is fair enough. A proper system for storing name/pair values and transferring those across the serial protocol makes a lto more sense. Heck, you could even get rid of the CLI, coz that's really all it's ever used for..... (it's not like it's a linux command line exactly :) )

robustini, 2015-10-07:
News about this feature?
Really annoying, there's way to make a full backup without going through the CLI?

RyanEwen, 2016-06-21:
I was very surprised to find that the backup & restore feature doesn't include CLI changes. Please change this.

hydra, 2016-07-23:
Current backup/restore functionality will be replaced with parameter group backup/restore in due course.  Firmware changes needed first.

closed 2017-02-25 by hydra