cleanflight/cleanflight-configurator #203: SERVO_TILT activation cause motors to start

sebmatton, 2015-05-29:
Hi all,

I was working with the cleanflight configurator today and was very surprised with this issue.

I did not changed anything but checked the "servo_tilt" in the configuration then clicked "save and reboot". When the board rebooted, the motors begun running !!! As if it was suddently armed. I don't think this is normal so maybe there is a problem with this...

Hope I can help doing something.

Thanks

sebmatton, 2015-05-31:
The explaination to this problem is the following : 
When servo_tilt is enabled, the motors 3 and 4 outputs are replaced with the gimbal values (default = 1500, so the motors are running). 

I still need to check if it is normal behaviour (or not), but it would be needed to have a warning when the user checks that checkbox (to prevent him to have the same issue as I had).

MauroMombelli, 2015-06-17:
This is because the "use servo" flag get activated, so everything that use servo will enable it. (that is an HW limitation of the timers-pin combination used with NAZE). Note that this change based on what FC you are using, and actually there is no way to support custom message based on FC recognized; for sure it can be discussed and can become a nice feature

Right know playing with option without knowing what you are touching with battery connected is NOT a good idea.

skaman82, 2016-06-16:
Placed a warning in https://github.com/cleanflight/cleanflight-configurator/commit/f890ee4d9262bd13083afad0e1a0c7e0799b83bb

closed 2016-06-16 by skaman82