cleanflight/cleanflight-configurator #206: Flash CC3D boards with OpenPilot bootloader

fennb, 2015-06-04:
Hey team,

With the mass proliferation of cheap CC3D boards from a multitude of manufacturers, it'd be great to move away from cleanflight's Naze-centric past and be able to fully support flashing these directly from the configurator.

Unless my experience and my reading of https://github.com/cleanflight/cleanflight/blob/master/docs/Board%20-%20CC3D.md is wrong, CC3D boards are still (kinda) second-class citizens (from a configurator point of view).

Would it be possible to natively port the OpenPilot bootloader and use the native protocol to flash a CC3D board directly via USB?

I've had a very quick look at the source and it looks like there's a few different approaches that could work (USB via SSP or maybe DFU).

I'm a developer but it's a bit beyond my current area of expertise - I'd love to help out/happy to dig into the internals if someone wants to shove me gently in the right direction/comment on feasibility.

Cheers!

tracernz, 2015-06-14:
May be useful: https://github.com/TauLabs/TauLabs/blob/next/flight/Doc/Architecture/dfu.txt

fftunes, 2015-08-26:
Would be awesome, ty.

closed 2017-02-25 by hydra