cleanflight/cleanflight-configurator #208: Servo Config Does Not Save

SugarBombs, 2015-06-06:
The config for the servos in the Servos tab does not save.

When saving, a data sending timeout occurs. 

In file msp.js, around line 1038 : case MSP_codes.MSP_SET_SERVO_CONF:

SERVO_CONFIG.length has a value of 10, likely needs to be max of eight.

After changing this line: for (var i = 0; i < SERVO_CONFIG.length; i++) {

To this line: for (var i = 0; i < SERVO_CONFIG.length - 2; i++) {

The timeout no longer occurs and the data is correctly saved.

SteveAmor, 2015-06-15:
Fixed in https://github.com/cleanflight/cleanflight/commit/9a8a31676b8612e924964f212da1a5c1e63f5c20 for Cleanflight code.

hydra, 2015-07-13:
Fixed in development branch.

closed 2015-07-13 by hydra