cleanflight/cleanflight-configurator #210: custom mix animation on setup tab looks strange in new version

kaefert, 2015-06-12:
Hi there!

I'm using a custom motor mix because my first motor pin died, so I fly my quadcopter using motor outputs 2 to 5 instead of 1 to 4. (see also https://github.com/cleanflight/cleanflight/issues/882 )

In previous configurator versions, I saw a simple rectlinear "box" with 6 labeled (front, left, ...) sides.
In the current version: `Running - OS: Linux, Chrome: 43.0.2357.81, Configurator: 0.65.0`
* the shape of this box seems to change on movement, and every of the previous 6 sides seems to be split into 2 along its diagonal into 2 same sized triangles.
* the labels are distorted.

[image]

DanNixon, 2016-01-13:
It looks like the texture coordinates are not being corrected for the perspective correctly (texture coordinates should be divided by the depth before interpolation).

skaman82, 2016-06-16:
solved in https://github.com/cleanflight/cleanflight-configurator/commit/a133eb63749b0109e226b326c41f448be65997d3

closed 2016-06-16 by skaman82