cleanflight/cleanflight-configurator #213: Built in documentation/help

stylr82, 2015-06-18:
Been giving some thought to adding built in help into the configurator. Basicly the concept is to have help icon next to each feature heading (e.g. Board Alignment)  When the user clicks the icon they are given a brief overview on what the each of parameters do, then perhaps have links off to the manual/documentation for more detailed info.

Have done a quick mockup of my idea - Just using jquery-ui to display a dialog. Just wanted to get some feedback to see if the idea is worth pursuing?

[image]

Also had a couple of thoughts around pulling in text from the .md docs and parsing text from them, but the docs would need to be consistently structured/titled for this to work effectively.

borisbstyle, 2015-06-19:
Looks good!

tracernz, 2015-06-19:
Excellent idea. :+1:

JohnOCFII, 2015-06-20:
Great idea!  If you have the skills to add it, I'm sure others will chime in with some of the more esoteric content.

thenickdude, 2015-06-24:
Yes this is very badly needed, especially for variables like `min_throttle` which everybody seems to think is related to their throttle lever on their Tx.

stylr82, 2015-06-24:
Ok, so I think this is worth perusing, I'll start work on this

stylr82, 2015-07-08:
I've done a pull request for this now, if anyone is interested in having a look. That's the easy bit though, writing up the various dialogs is going to take a bit of time

https://github.com/cleanflight/cleanflight-configurator/pull/217

skaman82, 2015-12-04:
This feature is already implemented in v 1.0

closed 2016-01-07 by skaman82