cleanflight/cleanflight-configurator #215: Config restore from 1.8.1 to 1.9.0 issue

rsappia, 2015-06-22:
Hi,

I was not able to restore my configurations from 1.8.1 to 1.9.0. I see that when I select the restore file, a message is displayed "Migrating configuration file generated by configurator: 0.65.0" and the packet error counter will start going high. Any suggestion or idea ? Am I missing something here?
I observed this issue when working with a CC3D flight controller.

thenickdude, 2015-06-24:
Can you post the contents of your original backup file please?

rsappia, 2015-06-24:
hi @thenickdude how can I upload the txt config file? I tried drag and drop but it wont work :(

thenickdude, 2015-06-25:
You can just paste its contents here instead of uploading the file separately. Or you can rename it to .jpg in order to trick Github into attaching it (they support attaching images).

rsappia, 2015-06-27:
Thanks @thenickdude , I have uploaded the file as .jpg
[image]

Davhed, 2015-07-06:
I'm having this problem too, but a little different. At first I was simply not bale to import backups from 1.8.1 to 1.9, but now I'm finding that I can't even restore from the backups I made using 1.9 and the new configurator (0.65).

I get this message:
`Migrating configuration file generated by configurator: 0.65.0`

Here is the contents of my latest backup using 1.9 and the Configurator 0.65:

````javascript
{  
   "generatedBy":"0.65.0",
   "apiVersion":"1.10.0",
   "profiles":[  
      {  
         "PID":{  
            "controller":2
         },
         "PIDs":[  
            [  
               1.3,
               0.037,
               17
            ],
            [  
               2.8,
               0.043,
               20
            ],
            [  
               6.5,
               0.063,
               5
            ],
            [  
               5,
               0,
               0
            ],
            [  
               0.15,
               0,
               0
            ],
            [  
               3.4,
               0.14,
               0.053
            ],
            [  
               5.6,
               0.33,
               0.083
            ],
            [  
               5.5,
               0.127,
               100
            ],
            [  
               4,
               0,
               0
            ],
            [  
               12,
               0.045,
               1
            ]
         ],
         "RC":{  
            "RC_RATE":1,
[... 1625 lines cut ...]
         "x":0,
         "y":0,
         "color":0
      },
      {  
         "directions":[  

         ],
         "functions":[  

         ],
         "x":0,
         "y":0,
         "color":0
      }
   ],
   "FC_CONFIG":{  
      "loopTime":3500
   },
   "ARMING_CONFIG":{  
      "auto_disarm_delay":5,
      "disarm_kill_switch":1
   }
}
````

newlakec, 2015-09-23:
I came across the same problem. My decision, reduced the backup file name and tried to restore, all is well.

closed 2017-02-25 by hydra