cleanflight/cleanflight-configurator #219: Luxfloat Needs more decimals for tuning

borisbstyle, 2015-07-18:


zenman8269, 2015-07-19:
So, does that mean you can use them in cli for the p term? In other words instead of 0.8 , 0.7, etc, use 0.85, 0.86, etc?

borisbstyle, 2015-07-19:
P supports up to 3 decimals
You can also have 0.852 for P.
But I and D can also have more as well.

zenman8269, 2015-07-19:
ok...I didn't know that. I have not been able to tune out some oscillations with roll p at 0.8 and 0.7 feels too loose. Will have to give 0.750 a try. Thanks :)

borisbstyle, 2015-07-19:
we are tuning in big steps now.

comparing p to PID1:
tuning step for p of 0.1 in luxfloat is same like 3.2 increase in pid1.

zenman8269, 2015-07-19:
So what about D term? It looks like in lux 18 is 0.018,  can you add decimals to it?
D term is not so bad, and I therm steps are quite small in magnitude, but it is definitely needed for P.

borisbstyle, 2015-07-19:
I think the values should be as they really are. 0.018 should not be 18.

I will see to modify this soon if someone else doesnt.

zenman8269, 2015-07-19:
It looks like if you set p_rollf = 0.750 by CLI, PID tuning tab still shows 0.8 and any subsequent saves will not effect the 0.750 setting. That is good :)
Subsequent saves in EZ Gui does change it back to 0.8 though. :(
That's going to be annoying!

borisbstyle, 2015-07-19:
I checked the code and this isn't straight forward as things are being multiplied from the MSP etc....

Pff .......who did ever chose to do it this way

Yamaford, 2015-07-30:
So inflight tuning steps will need an option to count in smaller steps now too.

borisbstyle, 2015-07-30:
yes for luxfloat it may be needed

Yamaford, 2015-11-13:
Why didn't this make it into the new version?

tracernz, 2015-11-13:
The way open-source works is if you want something you identify it(done with this issue), write the code, test it, and submit a patch. Nobody did the last 3 steps. ;)

Yamaford, 2015-11-13:
Actually it has been written and I've been using it for a couple months now but I couldn't find the PR . I see Boris closed it for some reason. https://github.com/cleanflight/cleanflight-configurator/pull/223

tracernz, 2015-11-13:
Sorry. For whatever reason the commit(s) didn't mention this issue then..

https://github.com/cleanflight/cleanflight-configurator/issues/219#issuecomment-156322747

LVNeptune, 2016-01-02:
I'm confused because the current configurator still only shows one decimal place for P. Am I missing something?

[image]

borisbstyle, 2016-01-02:
It is still not implemented

LVNeptune, 2016-01-02:
Why is @Yamaford claiming he is using it then..?

borisbstyle, 2016-01-02:
he used my outdated beta old configurator version. I believe hydra wanted a
cleaner approach for that.
Nobody ever picked it up though. But there are some new discussions going
on to solve it.

waltr1, 2016-04-11:
The latest BetaFlight firmware (2.6.x) has changed the LUX PID scaling.
This should take care of the problem opened in this issue.
See the ßF Wiki and Release Notes.

skaman82, 2016-06-16:
closing this since luxfloat scaling has changed

closed 2016-06-16 by skaman82