cleanflight/cleanflight-configurator #226: RCRate >=1.13 not saved correctly

bexamous, 2015-07-30:
In Receiver tab set RC Rate to 1.15, hit Save and then Refresh, value is now 1.14.  Hit Save and then Refresh and now it is 1.13... again and 1.12.  You can now Save/Refresh and it'll stay at 1.12.

Issue is this:

    $ RC_tuning
    Object {RC_RATE: 1.14, RC_EXPO: 0.65, roll_pitch_rate: 0, roll_rate: 0.7, pitch_rate: 0.7…}
    $ RC_tuning.RC_RATE
    1.14
    $ parseInt(RC_tuning.RC_RATE * 100)
    113

Problem is this:

    $ 1.14 * 100
    113.99999999999999
    $ parseInt(1.14 * 100)
    113

Probably want Math.round or something:

    $ Math.round(RC_tuning.RC_RATE * 100)
    114

This might be an issue elsewhere, I just always notice it with RC Rate.

fftunes, 2015-08-14:
I noticed in Luxfloat setting yaw rate = 0.29 gets "rounded" to 0.28.

thenickdude, 2015-08-20:
Indeed, test code:

```js
for (var i = 0; i < 1000; i++) {
   if (parseInt((i / 100) * 100) != i) {
       console.log((i / 100).toFixed(2) + " would save as " + (parseInt((i / 100) * 100) / 100).toFixed(2));
   }
}
```

Excerpt of results:

```
 0.57 would save as 0.56
 0.58 would save as 0.57
 1.13 would save as 1.12
 1.14 would save as 1.13
 1.15 would save as 1.14
 1.16 would save as 1.15
 2.01 would save as 2.00
 2.03 would save as 2.02
 2.05 would save as 2.04
 2.07 would save as 2.06
 2.26 would save as 2.25
 2.28 would save as 2.27
 2.30 would save as 2.29
 2.32 would save as 2.31
 2.51 would save as 2.50
 2.53 would save as 2.52
 2.55 would save as 2.54
 ...
```

Using Math.round() instead of parseInt() solves the issue and (using that same test instrument) all those values encode correctly.

There is a separate bug in Cleanflight which prevents PIDs greater than 2.50 from round-tripping (it clamps the values to 2.50 instead of the 2.55 that the byte could support).

hydra, 2015-10-14:
fix merged.

PR #237 (2015-10-14): Fix rounding of floating point values in crunch() (rc_rate, pids, etc)
Fixes #226 by properly rounding floating point values as they are scaled and stored as integers in msp.js's `crunch()`.
closed 2015-10-14 by hydra