cleanflight/cleanflight-configurator #227: 3D model on the main tab should show a more appropriate model for custom mixes.

NullVoxPopuli, 2015-08-02:
for example, if your custom mix has 4 motors setup, use the quad 3d model.

currently, it shows a REALLY big box.

thenickdude, 2015-08-02:
I don't think this is helpful as a craft with 4 motors defined needn't look anything like the default quadcopter model. It's also common for people to add additional motors to a mix to avoid burned out motor pins, for example.

The box does the job of showing orientation just fine, which is all it needs to do.

skaman82, 2016-03-15:
Can we just drop the ugly box and show the default quad model? The box doesn't do anything better than the default quad model.

skaman82, 2016-06-16:
see https://github.com/cleanflight/cleanflight-configurator/issues/210#issuecomment-226536678

closed 2016-06-16 by skaman82