cleanflight/cleanflight-configurator #230: feature request "Simple Configuration"

Ark0N, 2015-08-11:
Hi Dominic,

I have an idea for Cleanflight that could be a game changer. Especially for beginners. Add a new tab to cleanflight like the "Configuration" tab. Lets call it "Simple Configuration". 

There you have preconfigured settings for the hardware. An example would be the receiver.You got a list of receivers you can choose from. One of them is the Spektrum Sat(DSM2). When I choose them in the cleanflight "Simple Configuration" tab it tells me:

On Naze32, connect the signal pin to input channel 4. Solder the power to the naze 3,3v port. Connect the sat ground to any ground. (with pictures provided at best) <- Thats nice to have but not the killer feature thats next ->

Then cleanflight automatic configures the following fully automatic if I chosen the Spektrum Sat:

1. In ports tab, it sets UART2 to Serial RX.
2. In the configuration tab, it sets the receiver mode to RX_SERIAL.
3. It sets the steps to 1024 
4. Finally at the receiver tab, it map schannels to TAER1234

You could also add a button at "Simple Configuration" menu for binding the spektrum sat. Just make a button that releases the CLI command: "set spektrum_sat_bind = 5"

So a beginner can choose at the "Simple Configuration" just the spektrum sat and all settings will be made for him. And he even can bind it with a simple button. And he gets a guide how to connect it to the naze!

That was just an example for understanding. Into the future it would be very nice if I just choose all the hardware details of my quad at the "Simple Configuration". I choose my Frame, Motors, ESC, Props, Weight and it does the (nearly) best PID`s and all the other configuration for this hardware setup. This would make the configuration so much easier.

skaman82, 2016-06-16:
sort of duplicate of: https://github.com/cleanflight/cleanflight-configurator/issues/98

gaelj, 2016-06-17:
Wiring diagrams of each board is a good idea and has already been requested.
I think it should be separated from a configuration wizard though.
CLI commands can only be passed in CLI mode. 
Tabs use MSP commands only.
What we need first is to get rid of the CLI by introducing a generic get/set MSP command. I believe this is on its way already.
Then we need to define all pieces of hardware and how they can be mixed together. 
It's a nice idea but IMO not high priority.
The wiring diagram of the board that's currently connected, now that's a very nice feature on the other hand (I miss it sometimes)

gaelj, 2016-06-17:
PIDs cannot be guessed. Too many parameters in play

closed 2017-02-25 by hydra