cleanflight/cleanflight-configurator #233: Danger: 3D & Oneshot & Motor Tab safety checkbox

fftunes, 2015-08-15:
Sorry for double posting this, it's also here: https://github.com/cleanflight/cleanflight/issues/1208
...since i have no clue where it really belongs to.

With 3D & Oneshot enabled, clicking the safety checkbox makes the throttle signal jump from 1500 to 1000 (full reverse).

The checkbox does not work with Oneshot either way, true? If so, why?

leothehuman, 2015-08-19:
It works with oneshot, AFAIK. At least I'm able to use it to test motor rotation. Did you move the sliders to the middle position before checking the box?

daltonmatos, 2015-08-19:
Hello @leothehuman, I'm using configurator ``0.65.0`` and it's not possible to move the sliders before clicking the safety checkbox. I think that [cleanflight/cleanflight#1095] fixed this.

Although would be great if the configurator could change the value of the sliders to be consistent with the 3D feature, that is, move the throttle to ``3d_neutral`` value if 3D is enabled (don't know if it already does this after the mentioned fix, didn't test. =D).

fftunes, 2015-08-23:
@leothehuman , the sliders are by default in middle position. The safety checkbox is always checked. When thinking "Oh, let's uncheck it to be safe again... (klick)", they jump to full reverse.

hydra, 2015-12-16:
just has a look, this appears fixed now.  I spotted another problem with loading the 3D configuration settings too.

closed 2015-12-16 by hydra