cleanflight/cleanflight-configurator #234: jeti receiver not recognised in Clean Flight Configurator for use with Naze ImmersionRC Vortex

Oldhelidog, 2015-08-16:
Please look into Jeti Receiver not being recognized in CFC. 
Naze flight controller used in Immersion RC Vortex.
Using Jeti DS-16 everything works fine during initial Immersion RC tool setup.
Jeti DS-16 inputs are recognized and setup works as planned during Radio input mapping during Vortex on screen setup.   
CFCT will not recognize RX inputs for final setup in clean flight Config  .

There are know settings for Jeti RX in the { Serial Receiver Provider } box..

hydra, 2015-12-16:
You bug report is not very clear.  Please can you describe the problem you are having better as I don't know if this is an actual bug, feature request or support issue.

Marking it as support.  Suggest asking for help in the vortex or rcgroups forums and if there is still a bug then please give further details.

Likely this is nothing to do with the configurator either, it's probably something to do with the firmware in which case this issue is invalid and should be posted in the right place (github cleanflight issue tracker, not github cleanflight configurator issue tracker).

closed 2015-12-16 by hydra

Oldhelidog, 2015-12-16:
Updated to the newest firmware a month or so ago and it solved the issue..