cleanflight/cleanflight-configurator #235: New Tab: Failsafe

powdermnky007, 2015-08-17:
Would it be possible to add a failsafe tab that looks like a combination of the receiver page & mode page?  

It would let you set where each channel goes on the event of a failsafe and show you what mode the quad will be in (rate or angle).  It would also let you see if the buzzer is enabled during failsafe.  This would make it much easier to see, setup, and test failsafe settings.  

I crashed this weekend and my failsafe was set correctly, except that my buzzer didn't beep :(   so going to update that in the CLI tonight.  Never even crossed my mind.

ducktaperules, 2015-11-12:
I like this idea, with fail safe becoming more complex it makes sense that all of its features be accessible from a single place.

skaman82, 2015-12-04:
working on this right now

powdermnky007, 2015-12-04:
Fantastic!  I have a few suggestions I've been thinking about.  

1.  The main interface should be a drop down box with different presets.  For example, one of the presets would be:

Disarm and Cut Throttle: 
Useful for miniquads.  This preset will hold all stick values for a momentary loss of control, then on a full failsafe it will set throttle to zero and let it fall from the sky.  

Upon selecting that option, these cli values would be set. I got those CLI values from the writer of the failsafe routine, he posted them in the betaflight thread.  (cant remember if prodrone or digitalentity)
rxfail 0 h
rxfail 1 h
rxfail 2 h
rxfail 3 h
rxfail 4 h
set failsafe_delay = 10
set failsafe_off_delay = 0
set failsafe_throttle = 1000

2. Next to the preset box, put a checkmark or something saying "Show Advanced Features" 
that would then list all the failsafe cli values with a hover over description (like you did for the pid level settings).  Upon modifying any of these settings the drop down box would switch to custom?  However you want to handle that.

3.   If a serial rx is enabled (Frsky X4r) it should have a warning telling user to set radio to 'No Pulses'.  If CPPM/PWM is set, have a warning box that tells to consult their manual on how to do it.  and give a big red scary thing saying FAILSAFE WILL NOT WORK UNLESS YOU DO THIS!!!!!

4. at the bottom have a write up on how to test failsafe.  I know there are a few ways, but personally I go with the following method:
Remove Props
Plug battery into multirotor
Arm multirotor and throttle up until motors start spinning
Turn radio off
If motors stop and multirotor starts beeping you are done!
Note: Multirotor will not rearm until power its power is cycled

skaman82, 2015-12-04:
@powdermnky007 please have a look in here: https://github.com/skaman82/cleanflight-configurator/issues/45#issuecomment-161986993

skaman82, 2015-12-15:
done this

hydra, 2015-12-16:
merged

closed 2015-12-16 by hydra